jketterl/openwebrx #3: Allow for freq outside of defined band ranges

D0han, 2019-07-22:
> Sampling at 2400000 S/s.
Tuned to 148500000 Hz.
Tuner gain set to 7.10 dB.
Tuner error set to 43 ppm.
Reading samples in async mode...
client 0x1ace0f8: CS_THREAD_FINISHED, client_goto_source = 2, errno = 322019-07-22 02:54:10,413 - owrx.config - ERROR - 
Traceback (most recent call last):
  File "/home/openwebrx/openwebrx-jketterl/owrx/config.py", line 80, in fireCallbacks
    c.call(name, value)
  File "/home/openwebrx/openwebrx-jketterl/owrx/config.py", line 12, in call
    self.subscriber(*args, **kwargs)
  File "/home/openwebrx/openwebrx-jketterl/owrx/source.py", line 453, in set_dial_freq
    self.wsjtParser.setDialFrequency(self.localProps["center_freq"] + self.localProps["offset_freq"])
  File "/home/openwebrx/openwebrx-jketterl/owrx/wsjt.py", line 276, in setDialFrequency
    self.band = Bandplan.getSharedInstance().findBand(freq)
  File "/home/openwebrx/openwebrx-jketterl/owrx/bands.py", line 55, in findBand
    return next(band for band in self.bands if band.inBand(freq))
StopIteration
csdr shift_addition_cc: reinitialized to -0

Also maybe widen predefined ranges

jketterl, 2019-07-22:
I think you may have found a non-issue there. Basically what this means is that the system cannot determine a ham radio band (at least none of the ones configured in bands.json) from the currently tuned frequency. I think I have seen this before.

From my knowledge, the worst thing that's gonna happen from that is that you won't have band information on the popup on the map; it should still decode and show the messages.

I would however like to look into this since this seems to be a predictable scenario and I should probably catch that exception somewhere. It shouldn't reach config.py.

D0han, 2019-07-22:
Yeah, i do not claim this is breaking bug, i just think it could be caught and handled - tracebacks on expected output should not happen

jketterl, 2019-07-22:
agreed. i'll see what i can do :)

jketterl, 2019-07-28:
this should be fixed in commit accf2a34ff16e27d8dc409e46b4606def335614b

D0han, 2019-08-31:
Yep, that no longer appears, thanks!

closed 2019-08-31 by D0han