jketterl/openwebrx #5: Changing source work until freq change

D0han, 2019-08-31:
Steps to reproduce:
- load page with default source
- change freq by clicking on waterfall
- change source
- change freq again
At this point traceback is seen in console, and source can no longer be changed.
Reloading page brings some broken state, cannot be used anymore.

> csdr bandpass_fir_fft_cc: filter initialized, low_cut = -0.333333, high_cut = 0.333333
2019-09-01 01:21:45,637 - owrx.config - ERROR - [Errno 32] Broken pipe
Traceback (most recent call last):
  File "/home/openwebrx/openwebrx-jketterl/owrx/config.py", line 32, in setValue
    c.call(self.value)
  File "/home/openwebrx/openwebrx-jketterl/owrx/config.py", line 12, in call
    self.subscriber(*args, **kwargs)
  File "/home/openwebrx/openwebrx-jketterl/csdr.py", line 458, in set_offset_freq
    self.shift_pipe_file.flush()
BrokenPipeError: [Errno 32] Broken pipe

This was done on branch `develop`, `9bdeda78143ec75533e4af233766ad9402f56cf1`

jketterl, 2019-09-15:
I must admit that I have scratched my head quite a bit over this, and i haven't been able to reproduce it. the stack trace suggests that either the "csdr shift_addtion_cc" process has died, or has not been set up correctly after the source change, probably the latter, which is odd to me.
would you mind sharing the sources you have used? what i'd particularly like to know is if they have different sample rates, because the issue might be possibly arising when you switch from a higher sample rate to a lower one, which could potentially cause the shift to be out of band.
I'd also be interested if a specific timing is necessary in order to reproduce.

i have found that although the shell piping used extensively in openwebrx is very handy, it is highly susceptible to race conditions, and it also lacks some proper error handling when processes in the middle of the chain die. i will probably have to handle the processes and pipes better directly in python, but i'm a little scared that it might cause performance issues...

D0han, 2019-09-15:
All my sources are just profiles for one rtl-sdr. The steps i provided are distilled down to minimum and its stable reproduction for me.

```python
sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        "type": "rtl_sdr",
        "ppm": 43,
        # you can change this if you use an upconverter. formula is:
        # shown_center_freq = center_freq + lfo_offset
        # "lfo_offset": 0,
        "profiles": {
            "a": {
                "name": "a",
                "center_freq": 173000000,
                "rf_gain": 5,
                "samp_rate": 2400000,
                "start_freq": 173025000,
                "start_mod": "nfm"
            },
            "b": {
                "name": "b",
                "center_freq": 446000000,
                "rf_gain": 5,
                "samp_rate": 2400000,
                "start_freq": 446031250,
                "start_mod": "nfm"
            },
            "c": {
                "name": "c",
                "center_freq": 157000000,
                "rf_gain": 5,
                "samp_rate": 2400000,
                "start_freq": 157962500,
                "start_mod": "nfm"
            },
            "d": {
                "name": "d",
                "center_freq": 145250000,
                "rf_gain": 5,
                "samp_rate": 2400000,
                "start_freq": 145500000,
                "start_mod": "nfm"
            },
            "e": {
                "name": "e",
                "center_freq": 148500000,
                "rf_gain": 5,
                "samp_rate": 2400000,
                "start_freq": 148975000,
                "start_mod": "nfm"
            },
        },
    },
    # this one is just here to test feature detection
#    "test": {
#        "type": "test"
#    }
}
```

jketterl, 2019-09-27:
Ok sorry that it took me a while to figure this out, but I think I found the problem, or at least I have found a scenerio where I end up with the same stacktrace. Some explanation:

The standard version of csdr is missing the NFM deemphasis filter for 12 kHz. This filter would be used whenever you listen to NFM on openwebrx with a client that uses a 48 kHz audio sample rate (I have seen this happen with windows clients, mostly, it seems like windows defaults to 48 kHz when the hardware supports it).

When you first connect, openwebrx somehow fails at setting up the correct samplerate (I will need to look into that), so the audio works at first, even though it is running slightly to fast, and it will also encounter some underruns. As soon as you change the source, however, the sampling rate somehow gets through, and the pipeline set up at this point fails due to the missing filter.

The final step then brings up the stacktrace, since it tries to communicate with a pipe that is no longer there.

Now, if my assumptions are correct, there is a very simple fix to this: A while back, I did a fork of csdr to integrate exactly that missing filter: https://github.com/jketterl/csdr/tree/48khz_filter

Can you try that (please make sure you're using the "48khz_filter" branch) and see if it fixes the issue?

D0han, 2019-09-27:
Actually it was in Firefox on Linux. I think i also checked it in Firefox on Windows 10 (with a lot higher audio sample rate, maybe because of 5.1 speaker setup), but i do not remember now.  

Will check csdr when i have the chance. Thx!

jketterl, 2019-09-27:
Yeah I was able to reproduce it on Linux, too, but only after modifying the configuration. I don't have any reliable data on how likely this scenario is.

If it doesn't work, please let me know what sample rate you're running on (if it isn't 48 kHz). I can include other filters.

D0han, 2019-09-29:
So i checked with openwebrx 1b95807ac6975b56eb641d4a2c8b9147178b3d64 and csdr a87b7bb303f9448d7777cacadfe3e8b8e7bf2f27 and it seems that reproduction no longer works.

Thanks!

closed 2019-09-29 by D0han

jketterl, 2019-09-29:
happy to hear that. :)

i have opened #8 to cover the sample rate issues on initial load.

D0han, 2019-09-29:
Just checked with Firefox on windows where in log i see `Web Audio API succesfully initialized, sample rate: 192000 sps` - this also works fine (except for overfilled, red status bar, but whatever)

jketterl, 2019-09-29:
whoa. that's a lot. but, for whatever reason, there's a 48kHz (192kHz / 4) deemphasis filter available in csdr, and so it should be mostly fine. that filter however has been there before, so no idea why it failed.

i have read in the comments that the sample rate is not selectable, i.e. is dictated by the browser / os. i'll have to check if that is still current, i don't really think it's necessary to transfer NFM audio at dvd audio quality.

D0han, 2019-09-29:
Status bars list it as `Audio output`, not `Audio stream`. I assumed it pushes `Audio stream` amount thru network and decodes it to `Audio output` on client to satisfy all speaker channels - in that case that is not an issue.

jketterl, 2019-09-29:
i just had another look at the code, there's actually no constant resampling by 4. openwebrx tries to determine a factor to achieve a sampling rate between 8000 and 12000 Hz. For 192kHz, it picks factor 16, which then uses the new 12kHz filter. perfect :)