jketterl/openwebrx #6: Expose settings in URL

D0han, 2019-09-01:
This is a not an issue, its a feature request.

I think it would be really neat to expose current user settings in URL.  
This way, it would be super easy to share link with someone to specific transmission, or even make some bookmarks.

[Fragment identifier](https://en.wikipedia.org/wiki/Fragment_identifier) is a great candidate for such data, as it is not sent to server. Client code can freely read or change it without any interruption.

jketterl, 2019-09-01:
this is an interesting idea indeed, it may produce some lengthy urls though (not particularily worried though). from the documentation i know that openwebrx is supposed to accept at least some query parameters already, i am not sure which, and whether they still work, since i have never used that feature myself.

i have seen other receivers implement a "share" button that allows the current settings to be captured in a url, probably working together with that original feature.

there is probably a few things i need to fix before i can have a go at this, considering that right now even the defaults from the config are broken... at the moment i don't have a good concept for the frontend code yet. there's a lot of stuff going on that i need to understand first, too.

jketterl, 2020-01-25:
i had a shot on this today. it's not that easy to hook everything together correctly, the whole demodulator code is in dire need of an overhaul. i have tested it and it seems to be working so far.

jketterl, 2020-02-08:
since there's no immediate feedback, i'm closing this. feel free to open new issues if any bugs pop up.

closed 2020-02-08 by jketterl