jketterl/openwebrx #7: docker running error

d9394, 2019-09-27:
I put the docker file running on a synology nas machine ( cpu is intel atom c2538), so startup the contianer, after a few seconds , it's log get a error: Segmentation fault (core dumped)

here is the logs :
`542 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
541 - csdr - DEBUG - restarting since rc = 0, self.running = true, and no modification
541 - csdr - DEBUG - dsp thread ended with rc=0
nc: connect to 127.0.0.1 port 4950 (tcp) failed: Connection refused
525 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
524 - csdr - DEBUG - restarting since rc = 0, self.running = true, and no modification
524 - csdr - DEBUG - dsp thread ended with rc=0
nc: connect to 127.0.0.1 port 4950 (tcp) failed: Connection refused
511 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
510 - csdr - DEBUG - restarting since rc = 0, self.running = true, and no modification
509 - csdr - DEBUG - dsp thread ended with rc=0
nc: connect to 127.0.0.1 port 4950 (tcp) failed: Connection refused
496 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
495 - csdr - DEBUG - restarting since rc = 0, self.running = true, and no modification
494 - csdr - DEBUG - dsp thread ended with rc=0
491 - owrx.source - DEBUG - shut down with RC=1
Segmentation fault (core dumped)
nmux: (main thread/for) end input stream, exiting.
Connection to 127.0.0.1 4950 port [tcp/*] succeeded!
client 0x7f93920b6b60: started!
nmux: pthread_create() done, clients now: 2
173 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
client 0x7f93920b6ac0: started!
nmux: pthread_create() done, clients now: 1
nmux: listening on 127.0.0.1:4950
071 - owrx.source - INFO - Started rtl source: rx_sdr -F CF32 -s 500000 -f 14150000 -p 0 -g RFGR=4 -a \Antenna A\ - | nmux --bufsize 126976 --bufcnt 394 --port 4950 --address 127.0.0.1
064 - owrx.source - DEBUG - nmux_bufsize = 126976, nmux_bufcnt = 394
062 - owrx.source - DEBUG - Spectrum thread initialized successfully.
060 - owrx.bands - WARNING - Frequency for wspr on 4m is not within band limits: 70091000
055 - owrx.source - DEBUG - activating profile 20m
018 - owrx.connection - DEBUG - client connection intitialized
190 - owrx.source - INFO - SDR sources loaded. Availables SDRs: SDRPlay RSP2
189 - owrx.source - ERROR - The RTL source type \test\ is invalid. Please check your configuration
118 - owrx.source - ERROR - The RTL source type \rtl_sdr\ is not available. please check requirements.
Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>
_______________________________________________________________________________________________
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
`

jketterl, 2019-09-27:
Yeah, I've had this happen to me before, haven't been able to find a fix for it yet.

Since it's an atom CPU, I'm assuming it's the amd64 image. There's something broken with soapysdr or rx_sdr, I haven't figured it out. When it first happened, I tried out some older versions, and kept getting the segfault, so I'm guessing it's some incompatibility introduced with an alpine update.

Interestingly enough, the arm images seem to work fine...

I don't have much experience debugging segfaults, but if i got a minute, I'll check it out.

d9394, 2019-09-28:
Thanks for your reply. 
I just try to use another docker image : https://hub.docker.com/r/stanback/docker-rsptcpserver/ to test my soapysdr device , it seem's OK and working properly.

And I had tried your latest Raspberry Pi SD Card Image on raspberry 2B+, but it seem not work good: no FireFox browser , it seem something wrong:
[image]
[image]

d9394, 2019-09-28:
I find a bug in latest Raspberry Pi SD Card Image:
After the raspberry startup(or reboot), the rx_sdr process didnot run up, so I try the command manual:  
rx_sdr -F CF32 -s 500000 -f 14150000 -p 0 -g RFGR=4 -a \"Antenna A\" - | nmux --bufsize 126976 --bufcnt 394 --port 4950 --address 127.0.0.1
then system return:
Failed to activate stream
so I modify command:
rx_sdr -F CF32 -s 500000 -f 14150000 -p 0 -a \"Antenna A\" - | nmux --bufsize 126976 --bufcnt 394 --port 4950 --address 127.0.0.1
it run OK

jketterl, 2019-09-28:
Well, from your screenshots I can now tell you are running an RSP1, whereas the config in the repository (which is included as default config on the Raspberry image) happens to be for an RSP2. The first thing I noticed is that the ports seem to be labeled differently: yours seems to have a single port called "RX", the RSP2 has "Antenna A" and "Antenna B", so I'd suggest editing that in the config.

From your latest post I see that you left out the gain setting, even though, again in the screenshot, rx_sdr is reporting that it supports the RFGR setting. I am suspecting that the gain value is out of range. I have never owned an RSP1, but I read elsewhere that it supports 0-3. I have achieved best results so far by setting this to 0, so I'd suggest you try that first.

d9394, 2019-09-28:
yes, I already found the config_webrx.py has item : gf_gain , it can modify to 0 to run normal properly.

jketterl, 2019-09-28:
excellent, i will probably adapt the default config to use 0 as a value. it seems to work with all sdrplay devices.

are you by any chance able to test the same settings in the docker image? it uses the same default config, but you can mount a volume als "/config" to be able to edit it.

d9394, 2019-09-30:
I had tried to  modify gf_gain in config_webrx.py while runing docker image, it also not working, the error message is "Segmentation fault (core dumped)". 

I will try to  using your dockerfile to build a docker image on my NAS in the future days.

jketterl, 2019-09-30:
OK, let me know how it goes. There's a script `build.sh` in the repository that you can use, It doesn't do much apart from setting a variable depending on wether you're on x86 or arm.

d9394, 2019-10-06:
I just test rebuild a docker image on my NAS system by using /openwebrx/docker/Dockerfiles/ , but it also not work on my NAS, the error as same as this top. I think it was some problem with alpine OS.

So , I'am tring to write a new dockerfile by use ubuntu:latest OS, and after I build image with it , it appears another problem: 
[image]

d9394, 2019-10-06:
OK, I fix the 'ascii' problem by edit /openwebrx/owrx/controllers.py:
[image]

jketterl, 2019-10-06:
Wow I'm impressed that it's so easy to swap out alpine with ubuntu. I had a pretty hard time going from debian to alpine. Now, the problem with ubuntu is that it produces very large images as compared to alpine. But it confirms my suspicions that something is wrong there, probably with the provided C libraries.

I have merged your pull request with the encoding. I guess most platforms nowadays have their default encoding set to utf-8, hence nobody stumbled across that problem. Also makes me wonder why ubuntu doesn't use utf-8.

D0han, 2019-10-06:
Personally i am advising people to NOT use alpine. Debian/ubuntu images are not that large, and often size of image does not really matter.
Alpine on the other hand is using musl instead of glibc and busybox instead of normal shell and userspace tools. This creates problems - sometimes visible ones like here, sometimes obscure and hidden. This means that software is developed, optimized and debugged on different software stack than after deploying to production - this makes no sense.

jketterl, 2019-10-06:
Well, in that case, I can only advise not to use docker at all (which would also be my usual recommendation) because I am not developping on docker :wink:

D0han, 2019-10-06:
Sure, docker is not for everything and everyone. It has flaws, lots of them. 
Still, sometimes can be useful. But for that, it needs to be used in rational way. One of the things is giving it execution environment (userspace etc) same (or as similar as possible) as during development. So glibc in specific version, not musl; socat in specific version; shell in specific version...

jketterl, 2019-10-06:
I would love to do so. unfortunately, there is one more problem with the sdrplay devices: their drivers are provided as binaries, without any explicit specification which linux or library versions it was built on / against. So it's pretty much trial and error. I have used them with alpine before and it worked, it just doesn't any more right now. The same could happen on ubuntu, debian, or pretty much any other base system.
If they'd provide their drivers as source code, the error would probable show up during compile time.

D0han, 2019-10-06:
I agree with the source point. If they provide binary blob for rpi, they probably target official, debian buster based system. For other i have no idea, i do not have their products.

jketterl, 2019-10-06:
ok so just an update on what i see so far. i have set up the debugger, and the segfault is coming from libmirsdrapi-rsp, which is indeed the driver library provided by sdrplay. the stacktraces are next to useless though, since they stripped the debug symbols (of course). I can also tell that there seems to be some unresolved symbols coming from that library, they look to be non-standard calls, but i'm not too sure on that. I tried some hacks to fill those in, but no luck. I don't think I can fix this, this looks like something that sdrplay needs to address...

closed 2019-10-24 by d9394

Gymnae, 2019-12-03:
I tried the docker image last night on an x86_64 Intel NUC system running CentOS 8 with permissive SELinux.
No SDR device is found and it throws a websocket error. I tried defining sdrplay and/or airspy device connectors with the same result.

jketterl, 2019-12-03:
sdrplay on x86 docker is still broken, i reached out to sdrplay, who wanted to send me an updated version of the driver, which I haven't received yet. I suppose they will bring the version 3 out of beta soon (given that the new rspdx only runs with it), i'll probably have a go some day.

as for airspy, it might be that the docker image does not have the sdrplay airspy module installed. i'll have a look.

jketterl, 2019-12-03:
I just pushed new docker images for x86 with airspy support. From what i have seen, the openwebrx:latest and openwebrx-full:latest may still be broken due to sdrplay. The one that should work now is openwebrx-airspy:latest.

It might work in the full image if you specify `"device": "driver=airspy"`, that's how i was able to get SoapySDRUtil to cooperate.

jketterl, 2019-12-14:
I heard back from SDRplay this week, supposedly their driver version 3 is now working correctly on alpine. I have opened #36 for the necessary work.