jketterl/openwebrx #14: Labels?

Lonecrowe, 2019-10-03:
Any chance you could come up with a way of attaching permanent labels to specific freqs similar to how kiwi does it.  Or differently.  Just something to identify known signals?  Since you seem to be the only person actively still developing this software?

jketterl, 2019-10-03:
I'm not sure if you've seen my recent changes, but I just introduced "something like that" last week. Here's a screenshot:

https://pbs.twimg.com/media/EFgzKZOXUAAlvT8?format=jpg&name=large

It is setup with a simple json on the server side, and users are also able to store their personal bookmarks in localstorage.

Not sure if that's what you're after, if not, could you give a more detailed description, please?

Lonecrowe, 2019-10-03:
Ooooooh.  NICE!  I mean ideally like the kiwi you can (if I recall) ctrl click or have a + button for the exact setting you are on or listening to and it would store that to the json.   Editing the file is handy for when you have a large list.  I literally have a spreadsheet of aviation frequencies I can pick up where I'm at that I'd like to monitor.  But for a casual user it could be handy to add from the gui too. 

Is this on any live servers I can see?

jketterl, 2019-10-03:
I am not too familiar with KiwiSDR and how they solved it. It wouldn't be too much of a technical problem to write data from the browser back to the server and have it stored there, but since there's no authentication in openwebrx, I didn't want to have that due to possible abuse. The json on the server should usually be out of reach for the public.

I'd like to build some sort of preferences / settings gui for the whole receiver, that could have some way of editing the labels, too. But I need to figure out a way to have a way of protecting these kind of things.

I don't have much feedback about who has updated to what version, but you can checkout one of my receivers:

http://sdr.justjakob.de/ is a RPi 3B+ with an rtl-sdr; it should be available 24/7
http://sdr2.justjakob.de/ accesses my development machine directly; it's an i7 with both an rtl-sdr and an sdrplay, but it may be unavailable and/or broken at time.

Lonecrowe, 2019-10-03:
Very nice.  I see I can add bookmarks manually too.  Do people see those? Can anyone add them?  That could be a problem if its a public server and someone wanted to be malicious.  If its just for the session then that would be fine.  Although the method of adding the bookmarks from the gui is nice especially if you could color code them.  

That is cool how you have the time slots separated and you can see who's talking.  

I'm waiting to buy something with a large band width and then I'm going to give this a shot running it.  I'll probably do the RPi too! Excellent.

jketterl, 2019-10-03:
No, the bookmarks you add are stored in your browser's localstorage, so only you can see them. They should follow the same format, though, so one future feature might be a simple "export" which allows you to simply add all the bookmarks in the receiver, and then just paste it into the config file.

The DMR timeslots? Yeah I was always disappointed how most tools wouldn't show you any of the additional information that the protocols have. So I built my own :smile: 

I'm loving the SDRPlay devices. But keep in mind that a Raspberry will not be able to take the large bandwidths. It is already struggling with the 2.4MS/s if you set up some background jobs, or if you have a bunch of users. There's not way it could take 10MS/s.

jketterl, 2019-10-04:
So, what do you say? I'd like to close this issue for now, simply because there's no action to be taken right now. I'd still like to hear whatever requests you'd like to see implemented, but I'd suggest you open new issues then. Sound good?

closed 2019-11-01 by jketterl