jketterl/openwebrx #17: feature request

d9394, 2019-10-06:
Can merge the mobile view code ?
such as : https://github.com/bclswl0827/webrx4mobile

jketterl, 2019-10-06:
I have seen it on the original project; unfortunately, the communication between the server and the client has been completely rewritten, so I doubt it is still a drop-in replacement. Also, I still think that it won't be very handy to use if you squeeze the whole spectrum down to a smarphone display... that needs some clever ideas, which I don't have right now...

d9394, 2019-10-06:
OK, I give up this idea.

closed 2019-10-06 by d9394

jketterl, 2019-10-07:
i'll put a mobile view on my todo list. i really think it would be a good addition to the project, even though right now i can only imagine using it on a tablet. i'm always open for ideas on how such a mobile view would look or work like :)