jketterl/openwebrx #18: another bug?

d9394, 2019-10-06:
after I change band from browser some times( maybe so frequently or so fast ), the system crash:
[image]

jketterl, 2019-10-06:
the only thing i can see from the screenshot is that something is trying to send data, while the connection has been closed. there must be something before that.
have you checked if your sdr device is still available? It has happened to me many times that upon switching bands the usb device gets failed over by the kernel and cannot be used any more. There's not much I can do about that.

d9394, 2019-10-06:
yes , you are right: after I reconnect rsp device, this error was disappeared

jketterl, 2019-10-07:
i opened a separate ticket for better error handling. while it may not be possible to recover failed devices, the communication about what's going on could be improved.

also, just fyi: this has happened to me so often that i am now considering my own rtl_tcp fork that would change sdr parameters without releasing the device. :wink:

d9394, 2019-10-08:
I found the fact that when RSP device has error , the device's led display it was also inuse? maybe the usb port halt or some process holdon it ? So I must unplug and replug the usb cable to resolve.

jketterl, 2019-10-08:
yes, i have seen that on rare occassions that an sdr process keeps running in the background, even though it has reportedly been terminated. I suppose that that is due to the pipes used by openwebrx, since the process is usually "alone" (i.e. the corresponding nmux and csdr convert is gone).

I suspect that when the pipes are started and stopped in quick succession, the pipe itself already gets shut down, while the sdr process is still starting up and not handling the signals. I am unable to examine this any closer.

I do intend to re-write the way processes are chained in openwebrx, but that will be a bigger change, and I am concerned that handling the pipes in python may significantly increase the load.

d9394, 2019-10-22:
Yeah, now is running stabilize, thanks for working hard.

closed 2019-10-22 by d9394