jketterl/openwebrx #19: Handle failed sdr devices better

jketterl, 2019-10-07:
when an sdr device fails, connected users potentially enter a reconnect loop; the user interface is pretty much unusable, and other sdr devices that may still be functioning are inaccessible.

ideally, openwebrx would remove failed devices from the profiles dialog. if no devices are left, a message should appear that no sdr devices are availalbe.

also, users that are using an sdr at the moment it fails should receive a message about what happened, informing them to switch to another profile.

jketterl, 2019-10-07:
one case of this happening has been brought forward in #18

d9394, 2019-10-08:
That's a good idea

jketterl, 2019-10-13:
The latest version in develop has some patches to show a message in the websites' log panel if something goes wrong. it should also prevent the client/server reconnect loop. Of course, when I went to test it, all my sdrs worked like a charm... I'll have to run this for some time to see if there's anything I messed up or missed.

d9394, 2019-10-13:
Roger, I will build a new docker image for test new commit.

d9394, 2019-10-14:
I had build a new docker image ,and test it about an hour, that' great improve , very stable runing on my nas.

jketterl, 2019-11-22:
Here's my quick sum-up of what I gathered on this specific issue in the past few weeks:

- The error message is too easy to overlook, it even happens to me. I'd like to use the existing "press to start" overlay do display a modal message.
- It doesn't catch all error cases, sometimes its starts a reconnection loop together with the client, until the available connections are exhausted. The client still keeps looping then (separate issue)

commit 6f9ba6c2 (2019-12-21): improve sdr failure message display, closes #19
closed 2019-12-21 by jketterl