jketterl/openwebrx #21: Second client spawns new sdr proccess

takgr, 2019-10-22:
I've installed your fork of openwebrx to an Ubuntu server with a PlutoSDR. Everything works fine when a single client is connected, but when a second client enters via the web interface, a new sdr source process is spawned, requesting the same resource being used by the first client.
Instead I think just the nc command should be spawned, listening to the existing stream of iq data. At least this is what happens in the original openwebrx and works with multiple clients.

Please find below the log when the second client is being connected:

`2019-10-22 08:24:53,227 - owrx.connection - DEBUG - client connection intitialized
2019-10-22 08:24:53,292 - owrx.source - DEBUG - nmux_bufsize = 1003520, nmux_bufcnt = 50
2019-10-22 08:24:53,295 - owrx.source - INFO - Started rtl source: rx_sdr -N 1 -B 4000000 -F CF32 -d driver=plutosdr -s 4000000 -f 432000000 -p 0 -g RFGR=55 - | nmux --bufsize 1003520 --bufcnt 50 --port 4950 --address 127.0.0.1
nmux: pthread_create() done, clients now: 3
2019-10-22 08:24:53,296 - owrx.source - DEBUG - received onSdrAvailable, attempting DspSource restart
2019-10-22 08:24:53,296 - owrx.source - DEBUG - received onSdrAvailable, attempting DspSource restart
2019-10-22 08:24:53,297 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_140624277498512_shift_pipe | csdr fir_decimate_cc 362 0.00041436464088397796 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_140624277498512_bpf_pipe 0.02896 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_140624277498512_squelch_pipe --outfifo /tmp/openwebrx_pipe_140624277498512_smeter_pipe 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.0022425176332044 | csdr deemphasis_nfm_ff 11025 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
client 0x55eae66a8d90: started!
nmux: cannot bind() address to the socket: Address already in use
csdr squelch_and_smeter_cc: fifo control mode on
csdr bandpass_fir_fft_cc: fifo control mode on
2019-10-22 08:24:53,306 - owrx.source - DEBUG - adding new output of type audio
fir_decimate_cc: taps_length = 9653
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr shift_addition_cc: fifo control mode on
csdr squelch_and_smeter_cc: initial squelch level is 0
2019-10-22 08:24:53,313 - owrx.source - DEBUG - adding new output of type smeter
csdr shift_addition_cc: reinitialized to -0
nmux: pthread_create() done, clients now: 4
client 0x55eae66a8de0: started!
Connection to 127.0.0.1 4950 port [tcp/*] succeeded!
client 0x55eae66a8d90: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
(overlap_length = 138) = taps_length - 1
csdr bandpass_fir_fft_cc: filter initialized, low_cut = -0.362, high_cut = 0.362
ERROR: Unable to claim interface 5: -16
[ERROR] not device found.
`

jketterl, 2019-10-22:
That's a new one! :smile: I don't have a PlutoSDR, and you're the first one I know of that uses one, so I'm very happy to hear that it seems to work.

As for the bug: Internally, openwebrx uses a monitor thread to see if the sdr process is still running or has terminated. As long as this monitor thread is there, it should not spawn a second instance. Unfortunately I cannot tell from the log you posted, what has happened to the monitor, since all I can see is that it fails to set up the new source (understandably). Can you please post the log of what happens on connect of the first user?

takgr, 2019-10-23:
Yes, I forgot to mention about PlutoSDR. Since it is supported by SoapySDR framework and thus by rx_tools software, all I had to do is modify the command line in source.py to adapt Pluto.
I replaced SdrplaySource and my line is now like this:
rx_sdr -N 1 -B 4000000 -F CF32 -d driver=plutosdr -s {samp_rate} -f {center_freq} -p {ppm}"

I use darkstar007's fork of rx_tools that supports setting bandwith (https://github.com/darkstar007/rx_tools) but I entered the values manually in source.py file and it works!

Back to the supposed bug now, I pulled the latest source from git and now it doesn't happen at first. I saw that there were many changes in device handling, so this bug might have already been resolved by now. 
I tried it with 3 concurrent users, and there doesn't seem to be any issue, so I will spread the link to more users today and get back if I see any issues.

Thank you for this great software! My plan is to build a quad band HF SDR using a single USRP1 device that will decode simultaneously all digital and analog transmissions. I do this already with FT8 but matching with your fork of openwebrx would be awesome!

jketterl, 2019-10-25:
Happy to hear that it is working now. If it shows up again, just post the log here.

As for the Pluto: Since I know that it is working, I will add it as a separate type. That way, you won't have to edit the code any more :)

closed 2019-11-22 by jketterl