jketterl/openwebrx #22: owrx.wsjt - ERROR - failed to decode job

D0han, 2019-10-26:
I decided to take this for a spin with services and ft8.
RPI 3B seems to be underpowered for even one service (ft8, one worker) and client at the same time, as i am getting choppy sound.

Anyway, im seeing this in logs:
> 2019-10-26 18:35:00,008 - owrx.wsjt - WARNING - wsjt decoding queue overflow; dropping one file
> 2019-10-26 18:35:15,008 - owrx.wsjt - WARNING - wsjt decoding queue overflow; dropping one file
> 2019-10-26 18:35:22,870 - owrx.wsjt - ERROR - failed to decode job
> Traceback (most recent call last):
>   File "/home/openwebrx/openwebrx-jketterl/owrx/wsjt.py", line 34, in run
>     processor.decode(file)
>   File "/home/openwebrx/openwebrx-jketterl/owrx/wsjt.py", line 164, in decode
>     self.outputWriter.send(line)
>   File "/usr/lib/python3.5/multiprocessing/connection.py", line 204, in send
>     self._check_closed()
>   File "/usr/lib/python3.5/multiprocessing/connection.py", line 136, in _check_closed
>     raise OSError("handle is closed")
> OSError: handle is closed
>

jketterl, 2019-10-26:
Well, let me put it like that... There's better platforms to run that amount of decoding than a Raspberry Pi. Nevertheless, it should be able to handle it for one band. I am running a RPI 3B+ on 40m, and since it's running with only 500kHz of Bandwidth, it can decode FT8 and FT4, while I'm still listening. I can't make that receiver public, but you can see the results here: https://pskreporter.info/pskmap.html?preset&callsign=DD5JFK-2&timerange=10800&hidelight=1&showsnr=1

http://sdr.justjakob.de/ is just a RPI 3B+ too, and that does FT8 and APRS on 2m, while you can still have one user.

I have found that they are very sensitive to temperatures, and they need a proper supply voltage. You can check easily with `vcgencmd get_throttled`. You want to see `throttled=0x0`, anything else means your CPU has been throttled in some kind of way. Details here: https://github.com/raspberrypi/documentation/blob/JamesH65-patch-vcgencmd-vcdbg-docs/raspbian/applications/vcgencmd.md

Anyway, for the actual bug :grin: if you are seeing this on band changes, I am already aware. What happens is that the decoding queue gets processed, but the service that queued the job has been terminated in the meantime and therefore has shut down the receiving pipeline. The best way to solve this would probably be to keep the service running until its jobs have been processed. Another way would be to retract the jobs from the queue on shutdown.

I haven't seen any harm from this though, so I've treated this as low priority.

D0han, 2019-10-26:
I just stated underpoweredness of RPI in case someone searched for it, not for you to fix ;)  
Hm, actually it would be helpful for such cases to disable service handling when there is X or more users, too much CPU usage or whatever. I could get services in idle time, and normal use when interacting.

As for the bug - yeah, didnt seem to broke anything. I just reported it for sake of reporting broken stuff. Maybe you/someone willing to make PR know about it, maybe not :)

d9394, 2019-10-28:
I have another set device running openwebrx :  raspberry 3B+ and a sdrplay rsp1, I have tested 3clients online and it also running stablely.
[image]
[image]

jketterl, 2019-11-01:
Yes, as I said, the service handling is a first version, certainly not the last. I suppose there's not a single solution that fits all here.

My first idea was to let the OS handle it for me, that's why the wsjt-x decoders are running with nice=10, and that's also why there's a queue. The problem is that the SSB demodulators are taking up quite a bit of CPU, too. Throttling those however would lead to sample loss across all of them, resulting in very bad decodes. That would mean that we'd still be using lots of CPU, while receiving close to no output. That basically leads me to the conclusion that we'll need to handle the prioritization and downscaling in openwebrx itself.

Raspberry Pi throttling: This is not the first time this has come up, and while you are right in that I cannot fix it, I can still make it transparent. I have actually been thinking about adding a set of indicators somewhere that would show you when your voltage is dropping, or your cpu is overheating.

Also, since we are already chatting away here: I am looking for some kind of discussion platform where we could basically talk freely about this stuff, without everything being an "Issue". If you have any good ideas let me know :smile:

D0han, 2019-11-01:
The old and tested solution for discussions are mail lists and IRC. I think IRC would be suitable for this case, where we want to rapidly iterate on ideas.

Gymnae, 2019-12-03:
Regarding to your location of a discussion: IRC is nice, but the cool kids are on discord now. A openwebrx discord would be great, since they are so many unconnected activities. Discord has persistent chats and different channels, could be good for this project.

I just went ahead and created a discord server: https://discord.gg/z7bANs

(Although riot or matrix suit me better, since they are self-hosted and open source, the ease of reach of discord is great)

jketterl, 2022-06-17:
Haven't seen this in a while, I suppose the restructuring of the pipeline finally got rid of it. Closing.

closed 2022-06-17 by jketterl