jketterl/openwebrx #25: FM Demod not working properly

dj1an, 2019-11-01:
Hi, got an issue with FM Demodulation, all other Modes work fine:

`2019-11-01 10:41:42,247 - csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr shift_addition_cc --fifo /dev/shm/openwebrx_pipe_140155962312632_shift_pipe | csdr fir_decimate_cc 170 0.0008823529411764706 HAMMING | csdr bandpass_fir_fft_cc --fifo /dev/shm/openwebrx_pipe_140155962312632_bpf_pipe 0.0265625 HAMMING | csdr squelch_and_smeter_cc --fifo /dev/shm/openwebrx_pipe_140155962312632_squelch_pipe --outfifo /dev/shm/openwebrx_pipe_140155962312632_smeter_pipe 5 2 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.003921568627451 | csdr deemphasis_nfm_ff 12000 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8`

`2019-11-01 10:41:42,249 - owrx.source - DEBUG - adding new output of type audio
csdr shift_addition_cc: fifo control mode on
csdr deemphasis_nfm_ff: deemphasis_nfm_ff: invalid sample rate (this function works only with specific sample rates).
fir_decimate_cc: taps_length = 4533
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps`

**csdr deemphasis_nfm_ff: deemphasis_nfm_ff: invalid sample rate (this function works only with specific sample rates).**
seems to be the problem.

I am using newest csdr from git.

73 de DJ1AN

jketterl, 2019-11-01:
Hello, yes this is a known problem with csdr, it doesn't include the fm deemphasis filter for 12kHz, which is needed if your client runs at 48kHz audio. Please try to use this fork and let me know if this fixes your problems: https://github.com/jketterl/csdr/tree/48khz_filter

I will open a pull request to the original author.

dj1an, 2019-11-01:
Thanks, was not aware of this Problem.
Your Fork seems to fix this.

jketterl, 2019-11-01:
No worries. Happy I could help you.

closed 2019-11-01 by jketterl

ofadam, 2019-12-30:
Would you be able to include this fork in the instructions for initial setup? It doesn't look like the original creator of CSDR will ever pull it.

jketterl, 2019-12-30:
I have included it here: https://github.com/jketterl/openwebrx/wiki/Manual-Package-installation-(including-digital-voice)

did you follow some other instructions? i'd like to change it, but i suppose i have forgotten where it's linked.

ofadam, 2019-12-30:
I got to it via the readme (it's midway down the page, but linking to the original CSDR): https://github.com/jketterl/openwebrx/blob/develop/README.md

jketterl, 2019-12-30:
oh, good catch. it's even in there twice. will fix. thanks :)