jketterl/openwebrx #26: Second channel muted, but still sounding

DE8MSH, 2019-11-07:
Hi Jakob.

If I use DMR and I muted one channel while two stations are talking same time, the muted station is still hearable. 

I know you fixed it. But it still happens.

73 and thx!

jketterl, 2019-11-07:
did you update digiham recently? i did some work to prevent both channels playing at the same time, so i may have broken something there.

the other occasion i have seen this was when the client auto-reconnects, seems like the frontend and dsp get out of sync.

jketterl, 2019-11-22:
I am moving this issue over to the digiham project, it is probably an issue in the code there.

closed 2019-11-22 by jketterl

DE8MSH, 2019-11-23:
Hey. After updating digiham like this

https://github.com/jketterl/digiham/commit/d76a73a124619235e079eb261ec5742386b57fc3

it works again.