jketterl/openwebrx #31: High pitch noise

D0han, 2019-11-30:
I just noticed, there is pretty irritating high pitch noise (exactly 12000Hz according to app on phone).   Changing volume using OpenWebRX volume slider makes it quieter and louder, so it makes it clear culprit.

Checked on 2cfeb6b6d6eb690dacf494845f1439bd6bacb616, Win 10, Firefox 70.0.1, more than stereo speaker system (Audio output is around 192ksps).

jketterl, 2019-11-30:
this is going to be pretty hard to trace down... i don't have such hardware.

would you mind trying out a the following things for me: does the noise continue if you switch to a digital mode, e.g. DMR? if yes: does it play all the time, or only while a digital transmission is received?

just trying to narrow down if it comes from the server, or if it's in the webaudio code...

if it persists all the time, would you mind trying out a few of the examples here: https://webaudioapi.com/samples/ and watch if the tone is present there, too? if that is the case, openwebrx probably isn't at fault...

thanks :)

D0han, 2019-11-30:
So.. this will be a fun one. I cannot reproduce it anymore.  
I checked few examples you linked, no issues there.

Before i reported it i was on c7eb5c430c89efae9ca31af72230a1eaed968743, noticed it, upgraded, checked if it was still there and only then reported.  
I guess i will test more stuff when (if ever) it happens again.

jketterl, 2019-12-01:
ok now that's a really weird one then.

there's one more thing that might be the cause of this: i implemented the new AudioWorklet api for audio playback, which requires a secure connection (https) or localhost. you can tell which implementation is active by looking at the "Log" panel: "Web Audio API succesfully initialized, using ScriptProcessorNode API" vs. "AudioWorklet API".

You can use these two URLs that point to the same receiver to test if unsure:
AudioWorklets: https://sdr2.justjakob.de/
ScriptProcessor: http://sdr2.justjakob.de/

let me know if it makes any difference, please.

D0han, 2019-12-01:
Checked yours, everything fine.  
Checked mine, noise is back.
```
Server acknowledged WebSocket connection.
Audio stream is compressed.
FFT stream is compressed.
Web Audio API succesfully initialized, using ScriptProcessorNode API, sample rate: 192000 Hz
```

Switching to DMR makes it go away, switching back to FM restores it. Its continuous, on closed squelch. Stops when audio buffer is empty.

jketterl, 2019-12-01:
ok... that indicates that it's present in the audio stream coming from the server. might be that the nfm deemphasis filter i made for 48kHz does not work as intended. is it present on any of the other anlog modes, e.g. FM, SSB?

D0han, 2019-12-04:
It is present on FM, AM, LSB, USB, CW and all DIG i have (BPSK31, FT8, WSPR, JT65,  JT9, FT4).
It is NOT present on DMR, DStar, NXDN and YSF, but on those audio buffer stays flat 0.

jketterl, 2019-12-05:
well, that rules out the deemphasis filter, since that would only affect FM.

the reason why digital voice is not affected is probably because there's no audio being sent to the client as long as there's no signal being decoded. I am assuming you'd hear the noise there, too, if there was audio coming in.

So right now, my best guess would be that it's coming from the resampling process in the client, I am assuming that it's not correctly interpolating the audio data. That is one part of the software I have not touched yet, it's called sdrjs, it's a part of csdr compiled to javascript. I'd really like to replace it, but I haven't gotten around to work on an independent javascript resampling algorithm yet.

jketterl, 2020-01-06:
dusting this one off... i have been doing some work on the javascript dsp routines (more precisely: moved all to native javascript code). it should behave like before, for the most part, but maybe i sqashed this bug in the process. I'd be interested to hear if this still exists with today's build.

jketterl, 2020-02-09:
It is very faint (-67dB) but it's still there. I managed to capture it with audacity, below is a spectrum analysis screenshot. this was taken with squelch closed, so there shouldn't be any audio at all.

[image]

D0han, 2020-02-09:
Just checked on 216ede189cacd248f45395f45fba65d07af6b4d6 and i cannot reproduce it right now.

jketterl, 2020-03-13:
I have heard it myself today when using a Berhinger mixer as a sound device. Again, running in 12kHz, on all modes, but only when audio is playing (i.e. nothing when switching to digital and buffer is empty). It's pretty much inaudible as soon as even the faintest noise comes on, so pretty much when any signal is heard, even if it's silence from a very strong station nearby.

What was curious about it: it was changing with every reload, and after a few it was gone.

I am now completely confused as to where this would be coming from.