jketterl/openwebrx #32: Websocket not created

Gymnae, 2019-12-03:


jketterl, 2019-12-03:
the root cause lies in this little line:

`SoapySDRDevice_make failed`

which would indicate that something went wrong with the SDR device itself. Can you check if the RSP1 is listed if you run `SoapySDRUtil --probe`? With the RSP1, it has happened to me that the kernel has detected it as an `msi2500` device, and loaded the corresponding kernel module, which blocks soapy. If that's the case, then I'd recommend blacklisting that kernel module.

Gymnae, 2019-12-04:
This is the output of  `SoapySDRUtil --probe="driver=sdrplay"`

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
#
----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP1
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=1
  serial=B0001P0001

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

So that seems to work, but when I run the software, this happens:
```
$ python3 ./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>
Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>

    
2019-12-04 18:42:29,554 - owrx.source - INFO - SDR sources loaded. Availables SDRs: SDRPlay RSP1
2019-12-04 18:42:44,647 - owrx.connection - DEBUG - client connection intitialized
2019-12-04 18:42:44,656 - owrx.source - DEBUG - activating profile 70cm
2019-12-04 18:42:44,665 - owrx.source - INFO - Started rtl source: soapy_connector -p 4950 -c 49183 -s None -f 438800000 -g "0" -P 0 -a "None"
Tuner gain set to 0.00 dB.
IQ worker thread started
setting up control socket...
[... 42 lines cut ...]
  File "/home/sdr/openwebrx/jketterl/owrx/source.py", line 797, in __init__
    self.sdrSource.addClient(self)
  File "/home/sdr/openwebrx/jketterl/owrx/source.py", line 319, in addClient
    self.start()
  File "/home/sdr/openwebrx/jketterl/owrx/source.py", line 278, in start
    self.setState(SdrSource.STATE_FAILED if self.failed else SdrSource.STATE_RUNNING)
  File "/home/sdr/openwebrx/jketterl/owrx/source.py", line 358, in setState
    c.onStateChange(state)
  File "/home/sdr/openwebrx/jketterl/owrx/source.py", line 836, in onStateChange
    self.dsp.start()
  File "/home/sdr/openwebrx/jketterl/csdr.py", line 611, in start
    command_base = " | ".join(self.chain(self.demodulator))
  File "/home/sdr/openwebrx/jketterl/csdr.py", line 141, in chain
    if self.decimation > 1:
AttributeError: 'dsp' object has no attribute 'decimation'
----------------------------------------
2019-12-04 18:42:45,791 - owrx.connection - DEBUG - client connection intitialized
2019-12-04 18:42:45,795 - owrx.websocket - ERROR - ValueError while writing data
Traceback (most recent call last):
  File "/home/sdr/openwebrx/jketterl/owrx/websocket.py", line 131, in _sendBytes
    (_, write, _) = select.select([], [self.handler.wfile], [], 10)
ValueError: file descriptor cannot be a negative integer (-1)
2019-12-04 18:42:45,798 - owrx.websocket - DEBUG - websocket loop ended; shutting down
2019-12-04 18:42:45,798 - owrx.websocket - DEBUG - websocket loop ended; sending close frame
```

jketterl, 2019-12-04:
now we are mixing things up, seems like the airspy was selected:

```
SoapySDRDevice_setupStream failed: setupStream invalid format 'CS16' -- Only CF32 is supported by SoapyAirspyHF module.
```

if you have multiple sdrs connected, please use a "device" key containing a soapysdr query (eg `"device": "driver=sdrplay"`) on the configuration :)

jketterl, 2019-12-04:
also... it seems like you're missing a samplerate: `-s None`

Gymnae, 2019-12-04:
This is the relevant section form the `config_webrx.py`:
```
sdrs = {
    "sdrplay": {
        "name": "SDRPlay RSP1",
        "type": "sdrplay_connector",
        "ppm": 0,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 0,
                "samp_rate'": 500000,
                "start_freq": 439275000,
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },
}
```
How would I define  `"device": "driver=sdrplay"` in your added section?

Also, this is the output from `$ soapy_connector`
```
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action bind
Tuner gain set to 0.00 dB.
IQ worker thread started
SoapySDRDevice_setupStream failed: setupStream invalid format 'CS16' -- Only CF32 is supported by SoapyAirspyHF module.
socket setup complete, waiting for connections
```
The udev warnings I can't judge - both devices work when using `airspyhf_rx` and `play_sdr`directly. In the past, I found all attempts to use SoapySDR to be futile, thus I have two instances of openwebrx running, using those "native" binaries to access them.

jketterl, 2019-12-04:
with added device:

```
sdrs = {
    "sdrplay": {
        "name": "SDRPlay RSP1",
        "type": "sdrplay_connector",
        "ppm": 0,
        "device": "driver=sdrplay",
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 0,
                "samp_rate'": 500000,
                "start_freq": 439275000,
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },
}
```

I'm currently thinking if or if not it's a good idea to pass that per default; after all, you already specified that it's a `"type":"sdrplay_connector"`. I am currently unsure how i would merge it with other device selection paremters, e.g. a serial number.

as for the other software... I won't write any other connectors for sdrs supported by soapy. I have tested sdrplay RSP1s, RSP2s and the RSPduo, as well as the Airspy R2 and Airspy Mini. They all work perfectly fine. The fact that the Airspy HF+ is not currently working is not to blame on soapysdr, a separate ticket for that exists and will be adressed as time permits.

if you can't get your devices sorted, then please unplug the any unrelated sdrs while testing for this ticket. as soon as you have two or more supported devices connected, you will have to tell the system which one it is supposed to use, otherwise the error messages will be completely random and out of context.

Gymnae, 2019-12-05:
Ok, now I unplugged the airspyhf+ and the only SDR connected is the sdrplay rsp1.

The config for sdrdevices looks like this:
```
sdrs = {
    "sdrplay": {
        "name": "SDRPlay RSP1",
        "type": "sdrplay_connector",
        "ppm": 0,
        "device": "driver=sdrplay",
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 0,
                "samp_rate'": 500000,
                "start_freq": 439275000,
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },
}
```

soapy_connector by itself starts without and error:
```
$ soapy_connector 
Tuner gain set to 0.00 dB.
IQ worker thread started
socket setup complete, waiting for connections
[INFO] Using format CS16.
```

This is the result. The website opens, but a websocket error is displayed and no waterfall or profiles are displayed:
```
$ python3 ./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>
Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>

    
2019-12-05 10:23:13,906 - owrx.source - INFO - SDR sources loaded. Availables SDRs: SDRPlay RSP1
2019-12-05 10:23:17,415 - owrx.connection - DEBUG - client connection intitialized
2019-12-05 10:23:17,420 - owrx.source - DEBUG - activating profile 70cm
2019-12-05 10:23:17,429 - owrx.source - INFO - Started rtl source: soapy_connector -p 4950 -c 60999 -s None -f 438800000 -g "0" -P 0 -a "None" -d "driver=sdrplay"
Tuner gain set to 0.00 dB.
IQ worker thread started
setting up control socket...
socket setup complete, waiting for connections
[INFO] Using format CS16.
control socket started on 60999
client connection establised
2019-12-05 10:23:17,834 - owrx.source - DEBUG - opening control socket...
control connection established
2019-12-05 10:23:17,835 - owrx.source - DEBUG - received STATE_RUNNING, attempting DspSource restart
2019-12-05 10:23:17,836 - owrx.websocket - DEBUG - websocket loop ended; shutting down
[... 76 lines cut ...]
    controller(handler, request).handle_request()
  File "/home/sdr/openwebrx/jketterl/owrx/controllers.py", line 155, in handle_request
    conn.handle()
  File "/home/sdr/openwebrx/jketterl/owrx/websocket.py", line 154, in handle
    self.read_loop()
  File "/home/sdr/openwebrx/jketterl/owrx/websocket.py", line 202, in read_loop
    self.messageHandler.handleTextMessage(self, message)
  File "/home/sdr/openwebrx/jketterl/owrx/connection.py", line 330, in handleTextMessage
    client = OpenWebRxReceiverClient(conn)
  File "/home/sdr/openwebrx/jketterl/owrx/connection.py", line 88, in __init__
    self.setSdr()
  File "/home/sdr/openwebrx/jketterl/owrx/connection.py", line 168, in setSdr
    self.startDsp()
  File "/home/sdr/openwebrx/jketterl/owrx/connection.py", line 205, in startDsp
    self.dsp.start()
  File "/home/sdr/openwebrx/jketterl/owrx/source.py", line 803, in start
    self.dsp.start()
  File "/home/sdr/openwebrx/jketterl/csdr.py", line 611, in start
    command_base = " | ".join(self.chain(self.demodulator))
  File "/home/sdr/openwebrx/jketterl/csdr.py", line 141, in chain
    if self.decimation > 1:
AttributeError: 'dsp' object has no attribute 'decimation'
```

Sidenote: Closing openwebrtx with ctrl+c does not kill soapy_connector, so I need to kill soapy_connector by hand. I'll keep this in mind for a systemd service unit, because I use a script to change between spyserver and openwebrx for accessing airspy devices.

jketterl, 2019-12-05:
I just copied your configuration onto my dev machine, and this really does look like some weird bug at first. The problem is indeed that the sample rate does not get setup correctly, and I did get the same stack traces.

It is hard to spot, but there's a spurious apostrophe in the "samp_rate" key right here: `"samp_rate'": 500000,` - once I removed that, the receiver worked.

here's the fixed config:

```
sdrs = {
    "sdrplay": {
        "name": "SDRPlay RSP1",
        "type": "sdrplay_connector",
        "ppm": 0,
        "device": "driver=sdrplay",
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 439275000,
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },
}
```

Gymnae, 2019-12-05:
Well, that's a big fat d'oh!
How did that sneak in there? Well thanks for your debugging. I was about to blame CentOS 8, because that OS is locked down and missing a bunch of packages. 

But now it works with SDRPlay RSP1 and I can close this issue. `soapy_connector` also closes properly. Wohooo

closed 2019-12-05 by Gymnae