jketterl/openwebrx #33: No Digital Decode

dellacer1, 2019-12-10:
Hello my Name is Andy,

I don´t have the Digimods on the Website, can you help me what is wrong?
The add-ons are installed.
[image]

Andy

jketterl, 2019-12-10:
Hello Andy,

the most common cause of this is a missing dependency, most likely the `sox` application (I have missed that myself many times). You can easily check what's missing, just append `/features` to your receiver URL (e.g. http://openwebrx/features). That will bring up a feature report, which will show you what specific dependencies are missing.

dellacer1, 2019-12-11:
Thanks,
It works good.
Andy

closed 2019-12-11 by dellacer1