jketterl/openwebrx #34: Docker arm64 image ?

nmaster2042, 2019-12-12:
I'm using a khadas vim2 SBC with aarch64 processor.

Do you plan to make a suitable docker image for it ?

jketterl, 2019-12-12:
I don't have access to any comparable hardware right now. I am aware that there's new instance types available on EC2, I will check them out as soon as I find time. I have been thinking of using one of these machines as a build server.

vielmetti, 2019-12-14:
It may be the case that issues with building `csdr` on arm64 need to be resolved first, per https://github.com/F5OEO/rpitx/issues/173 and https://github.com/simonyiszk/csdr/issues/43

jketterl, 2019-12-14:
I was just googling around and about to buy an Odroid... until I learned that this is actually just a matter of getting another image for one of my Raspberry Pi 3/4. I'm downloading the latest Ubuntu 64bit right now. Let's see if it's really that easy :slightly_smiling_face:

jketterl, 2019-12-15:
Success! I did have to patch the csdr cpu detection around a bit (https://github.com/jketterl/csdr/tree/cpufeatures), i hope that there's no collateral damage from that, and the sdrplay support is broken (same as x86_64, it seems). But other than that, it seems to be working!

I have pushed it to the docker hub, it is included in the manifest, so `jketterl/openwebrx:latest` should work for aarch64 now :)

nmaster2042, 2019-12-15:
I confirm it's actually running on docker aarch64. Great thanks.
I'll make more tests.

closed 2019-12-15 by nmaster2042

jketterl, 2019-12-15:
awesome, thanks for the feedback :)