jketterl/openwebrx #35: Problem with Android Browser an Audio

dellacer1, 2019-12-12:
Hi,

I have a problem with Android and the server, if I am using the tablet or smartphone then I have no demodulation, with the PC no problem. I use Chrome and Firefox as browsers, both with the same error. I attached the terminal issue.

Andy
[Fehler_openwebrx.txt](https://github.com/jketterl/openwebrx/files/3957858/Fehler_openwebrx.txt)

jketterl, 2019-12-12:
Hi Andy,

from your logfile I can see that you have unfortunately run into one deficiency of the original "csdr" project, namely the lack of a FM deemphasis filter for 12kHz. The missing filter is available on my fork of the project here: https://github.com/jketterl/csdr

Please try installing the custom version and see if it fixes the problem.

I have also sent this fix to the author, see https://github.com/simonyiszk/csdr/pull/48

Hope I could help :slightly_smiling_face:

dellacer1, 2019-12-13:
Thanks  it works fine

closed 2019-12-13 by dellacer1