jketterl/openwebrx #36: Update to sdrplay driver version 3

jketterl, 2019-12-14:
The newer sdrplay devices (RSPDx) are only supported from version 3 and up. Also, version 3 is supposed to be working on alpine.

frspin, 2020-04-07:
Is correct a change in owrx/source/sdrplay.py file like this:

    def getDriver(self):
        results = SoapySDR.Device.enumerate()
        return results[0]["driver"]
#        return "sdrPlay"
#        return "sdrplay"

for using device name of API 2.13 or API 3.06?

Regards

Franco Spinelli
IW2DHW

jketterl, 2020-04-07:
I doubt that's a valid thing to do. First of all, there'se no `SoapySDR` binding imported, and it's not just a missing import: The bindings are not available on all supported platforms.

Second, this will always pick the driver for the first device, which might be using any driver (even completely different ones, e.g. airspy).

I'm not completely sure at this point if the driver string will change at all between version 2 and 3.

jketterl, 2020-05-24:
I am by now completely confused by the amount of sdrplay modules for soapy.

https://github.com/pothosware/SoapySDRPlay
https://github.com/SDRplay/SoapySDRPlay
https://github.com/fventuri/SoapySDRPlay

Right now, it seems the version on the SDRplay account seems to be receiving updates, so I guess we'll take that...

jketterl, 2020-05-24:
Good news: the version from SDRplay is a drop-in replacement, it doesn't require any changes to OpenWebRX itself. I have already noticed that some sample rates are not supported any more, so please double-check your config with SoapySDRUtil.

The new driver and the soapy module from SDRplay have now both been included in the docker builds; I have also ported the changes to openwebrx-pi-gen, but haven't tested them (I'm running out of SD cards :open_mouth:) yet.

You will need to update / replace the driver and SoapySDRPlay module manually for:
- Manual installations
- Raspberry Pi images (for now, the next image build should include the right drivers)
- Package installations

I will update the documentation in the Wiki.

I have no idea how if I'm supposed to have dual-tuner support at this point, or how it is supposed to work, but the RSPDx should work.

closed 2020-06-03 by jketterl