jketterl/openwebrx #37: Direct Sampling on RTL_SDR

sm4xas, 2019-12-26:
Hello!

Love your work!

Is there any way of using direct sampling on the RTL_SDR V3 ?
I tried searching the wiki and docs but could not find this.
In ha7ilm version you could use the rtl_sdr -D option for this.

73 de SM4XAS

jketterl, 2019-12-26:
as of right now, the answer is no. from what I can see, the -D flag is not available on the official rtl-sdr version, so this probably needs some modification. I will need to do some research, if it's in librtlsdr, this will probably be too messy to include, I don't want to start distributing my own packages of everything...

sm4xas, 2019-12-26:
I understand.
There is a fork of the rtl_sdr which has direct sampling included. See below link.

https://github.com/simonyiszk/openwebrx/wiki/Using-RTLSDR-in-direct-sampling-mode-with-OpenWebRX

Is there any way of reversing back to using the rtl_sdr ? I would like all your lovely digital decoding possibilities.

jketterl, 2019-12-26:
yeah i found that fork, too. as i expected, it is patching the lib itself, so that's a problem. if i start using the calls provided by this fork, it won't compile against the official version any more, and if i work with precompiler statements (not sure how, probably some detection necessary) you will need to recompile the owrx_connector. that might be a viable option, but it won't be available on the debian packages or docker.

i just removed the old non-connector sources this week, after the feedback on twitter was that the connectors are working for everybody 😀 they'd still work, i just wanted them out of the way for cleaning up, and since the connectors work and also have other advantages... well. the basic structure won't go away since there's still the hackrf using the same pattern.

jketterl, 2020-01-06:
I recently did some unrelated work with the connectors, and i found out that the sparky module for soapysdr has the "direct_sampling" flag. i couldn't try it out since i don't have any sticks that are modified for that purpose. do you think that would be an option? it seems to be included by default

sm4xas, 2020-01-06:
Hello!
Yes this might work. I did a quick Google and found that soapy should have support.
https://github.com/pothosware/SoapyRTLSDR/wiki
https://github.com/pothosware/SoapySDR/issues/29

I'd be happy to try it for you :)

73 de SM4XAS

jketterl, 2020-01-10:
I just added an alternate source for that purpose. You can activate by switching the sdr device to `"type": "rtl_sdr_soapy"`. This type supports the new config key `direct_sampling`, value is passed to soapy (mine says possible values are [0, 1, 2]).

It works with an unmodified stick, but I'm unable to confirm direct sampling works. It does something since I don't see any signals on my waterfall when I activate it :grin: 

side note: I haven't included the direct_sampling field into the runtime communication of the connector yet, so for now it will only work reliably when its set on the device level (that passes the parameter directly on startup). I could make it work on the profile level, if that makes any sense at all, I'm not sure at the moment.

Let me know if it works :slightly_smiling_face:

ofadam, 2020-01-14:
I'm having trouble getting this to work – owrx keeps saying that the RTL source type "rtl_sdr_soapy" isn't available. Could you explain how to modify the base config to make this work?

jketterl, 2020-01-14:
OpenWebRX does a check if you do have the requirements installed. You can get a list of the requirements if you append "/features" to your receiver URL, check for the "rtl_sdr_soapy" driver there. Did you install the SoapyRTLSDR module? It doesn't come preinstalled with the RPi images or docker images at the moment.

If you have working config for an "rtl_sdr" type, it should work with the "rtl_sdr_soapy" type as well. All you need to do to enable direct sampling is adding a `"direct_sampling": 2` key, where the value will probably depend on the specific modification of the stick (from what I know, it's "I only", "Q only", ???)

ofadam, 2020-01-14:
Can you only run "rtl_sdr" OR "rtl_sdr_soapy" (not both)? Or should I keep
using "rtl_sdr" for non-direct sampling ranges?

jketterl, 2020-01-14:
you'll have to decide for one or the other per device, you can't switch the type at runtime. "rtl_sdr_soapy" should however work for normal operations as well, i haven't tested it very much though.

mripv6, 2020-01-25:
Just a quick note, I was able to make the RTL_SDR V3 dongle to work on a Raspberry PI 4 Model B using the suggestions here. I compiled/installed https://github.com/rtlsdrblog/rtl-sdr-blog, first. Then, rebuilt SoapySDR. I also upgraded the firmware in the Raspberry PI 4. (sudo apt upgrade). rtl_test showed some errors (lost at least xx bytes). However, SoapySDRUtil --probe="rtl-sdr" did not show errors. The PI firmware upgrade was required because SoapySDRUtil was showing write_reg_fail errors. As a side note, I was able to use "digital_voice_unvoiced_quality = 2" on the PI 4

I added the direct_sampling key word as per Jacob's instruction. The value has to be '2'. 

 sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        "type": "rtl_sdr_soapy",
        "ppm": 0,
        "direct_sampling": 2,
        ....

The profile section looks like this:
 "profiles": {
            "40m": {
                "name": "40m Ham Band",
                "center_freq": 7100000,
                "rf_gain": 30,
                "samp_rate": 1024000,
                "start_freq": 7000000,
                "start_mod": "lsb",
            },
           .......

jketterl, 2020-01-25:
oh wow. so if i'm interpreting this correctly, it basically still requires a modified rtl_sdr library to work, right? i was afraid something like that would pop up.

mripv6, 2020-01-25:
I'm not sure. The one I used came from rtl-sdr blog. I'll go ahead and test
the official version later today. I recall one version only took -D without
a value.

mripv6, 2020-01-26:
Jacob - I did some testing with the osmocom rtl-sdr driver. I built
rtl-sdr library, SoapyRTLSDR, SoapySDRPlay and SoapySDR. Then I
configured openwebrx to use "rtl_sdr_soapy" under the rtlsdr section of
config_webrx.py. Then, I set up a profile to listen to the 40M ham band,
which requires direct sampling. I also included "direct_sampling": 2 at
the rtlsdr level in config_webrx.py.

I'm happy to report that it works. So I think direct sampling was added
into the osmocom driver.

Note, the -D flag is in rtl_power.c on the osmocom git site. However,
the -D flag doesn't take a value. (0,1 or 2). Same for rtl_power.c on
the blog site referenced earlier. This threw me for a loop because when
I tried to use a value, rx_power and rtl_power didn't work as expected.

Compling SoapyRTLSDR required cmake -DCMAKE_CXX_FLAGS=-latomic ../
Otherwise, the build complains about not finding latomic. I found a
comment where the authors were aware and attempting to fix.

I suppose all this will make it into the repositories at some point so
we don't have to build from scratch. So I don't think you need to be
concerned... at least for now!

AgriVision, 2020-02-27:
It works out of the box using the latest image (2020-02-20-OpenWebRX-full.zip)
By defining two SDR devices I can easily switch between shortwave direct sampling and normal mode:

sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        #"type": "rtl_sdr",
        "type": "rtl_sdr_soapy",
        "ppm": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
................................
            },
        },
    },
    "rtlsdrds": {
        "name": "RTL-SDR USB Stick Direct Sampling mode",
        "type": "rtl_sdr_soapy",
        "ppm": 0,
        "direct_sampling": 2,
        "profiles": {
            "40m": {
                "name": "40m",
                "center_freq": 7000000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 6000000,
                "start_mod": "lsb",
            },
........................
       },
    },
}

jketterl, 2020-02-27:
Thanks for the report, glad it works :smile:

One piece of advice though: As long as you don't have two physical SDRs connected, you probably shouldn't configure two devices in the config. The reason is that OpenWebRX will assume that those two devices are available *simultaneously* and attempt to serve them if requested. This will probably result in some nasty errrors. The `rtl_sdr_soapy` source should be able to handle the 70cm profile, you just need to set `direct_sampling` on the profile level. You'll need to set `"direct_sampling": 0` on the 70cm profile to disable it.

jketterl, 2020-02-27:
Reopening since switching between profiles does not work; the connector still needs work...

jketterl, 2020-02-29:
switching between direct sampling mode using profiles is now implemented in develop and the latest owrx_connector.

closed 2020-02-29 by jketterl