jketterl/openwebrx #41: ARM64 docker image: no sound

nmaster2042, 2019-12-30:
I'm using a Khadas vim2 SBC with amlogic S912 processor with docker.

Contrary to tests made with my main computer ubuntu 19.10 x86_64 docker, on the SBC I got no sound whatever modulation I chose, whereas everything seems to be ok (no errors). 

The only thing strange is in status, CPU usage stays at 0%.

Both PC and SBC are on the same local lan.

jketterl, 2019-12-30:
Well, the log shows no real problems (apart from the sdrplay one which i cannot fix, unfortunately). Are you getting an S-Meter readout, or waterfall? Can you try moving the squelch? (I've had the squelch stuck in closed for some times now, no idea why it happens, but moving it all the way across usually solves the problem).

I don't think it's the FFTW_MEASURE bug since the libs in the container should not be affected, but it's not explicitly disabled, sojust in case: please keep the receiver window open and wait for ~30 seconds. If there's no audio then, it's not that one.

nmaster2042, 2019-12-30:
Yes I have the waterfall and smeter reading see below.

[image]

Tried moving squelch with no luck for now.

Letting received for more than 30 second is the same: no sound.

nmaster2042, 2019-12-30:
I also tried with another browser Chromium, same result.

If I move volume button completely right (full) I get little noise but it doesn't change when I change modulation.

jketterl, 2019-12-31:
when i look at the screenshot, i can see that your tuned frequency is out of band. not sure if you tried tuning after that, but please try to do so if you haven't. also, try setting the `start_freq` in your profile to something in-range.

nmaster2042, 2019-12-31:
I made a lot of tests, so I may be wrong on some of them.

I made new tests, here is my sdr configuration:

sdrs = {
    "airspy": {
        "name": "Airspy",
        "type": "airspy",
        "ppm": 0,
        "profiles": {
            "VHF": {
                "name": "VHF",
                "center_freq": 146000000,
                "rf_gain": 10,
                "samp_rate": 2500000,
                "start_freq": 145800000,
                "start_mod": "nfm",
            },
        },
    },
}

And the corresponding screenshot:

[image]

Still no sound.

jketterl, 2019-12-31:
i'm out of ideas for the moment. as i said, i've had very quiet audio before, and usually it goes away when you move the squelch, but as i said, i don't know what's causing it. i will get back to this when i find out more.

nmaster2042, 2020-01-03:
Happy new year.

I just tried the docker image on another arm64 SBC: an odroid C2 with an amlogic S905x processor running the lastest armbian version.

With this setup, all is fine and I get sound.

It was for test purpose only because this SBC has less power and can't handle the default 10 Mbps sample rate of my airpsy contrary to my khadas.

I'm not a docker expert, the only software difference is the OS. On my khadas VIM2, it's ubuntu bionic witch is older than armbian.

On hardware side the difference is an amlogic S905x 4 cores processor against an amlogic S912 8 cores processor on khadas vim2.

Could it be a build flag on some software part of the image ?
Or something else, I don't know.

But we can see amr64 docker image is working.

nmaster2042, 2020-01-06:
Hi.

I found an newer os image for my SBC, armbian for khadas SBC and other amlogic processors.

It's a work in progress and not official armbian supported but it has newer kernel and base os components.

I burned it on a micro sdcard and installed docker on it.

And I have audio now.

So my issue doesn't seem to be directly related to openwebrx image.

I wanted to let you know, it may be useful for other users.

closed 2020-01-08 by nmaster2042