jketterl/openwebrx #42: Slew of errors upon running

ofadam, 2019-12-30:
Sadly, I had everything working yesterday and now am stuck with these errors when accessing via browser. I appreciate any help you may be able to give.
___

2019-12-30 09:24:44,909 - owrx.source - INFO - Started sdr source: rtl_connector -s 2400000 -P 0 -g 30 -f 438800000 -p 4950 -c 53451
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
IQ worker thread started
socket setup complete, waiting for connections
setting up control socket...
control socket started on 53451
2019-12-30 09:24:45,732 - owrx.source.connector - DEBUG - opening control socket...
client connection establised
control connection established
2019-12-30 09:24:45,734 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
2019-12-30 09:24:45,735 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_3059244108_shift_pipe | csdr fir_decimate_cc 217 0.0006912442396313364 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_3059244108_bpf_pipe 0.028933333333333332 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_3059244108_squelch_pipe --outfifo /tmp/openwebrx_pipe_3059244108_smeter_pipe 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.0031662434559077 | csdr deemphasis_nfm_ff 11025 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
csdr squelch_and_smeter_cc: fifo control mode on
csdr shift_addition_cc: fifo control mode on
Connection to 127.0.0.1 4950 port [tcp/*] succeeded!
2019-12-30 09:24:45,786 - owrx.dsp - DEBUG - adding new output of type audio
csdr bandpass_fir_fft_cc: fifo control mode on
csdr fractional_decimator_ff: fir_decimate_cc: taps_length = 5787
client connection establised
use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
(overlap_length = 138) = taps_length - 1
csdr shift_addition_cc: csdr squelch_and_smeter_cc: initial squelch level is 1e-15
reinitialized to -0
2019-12-30 09:24:45,815 - owrx.dsp - DEBUG - adding new output of type smeter
2019-12-30 09:24:45,826 - owrx.bands - WARNING - Frequency for wspr on 4m is not within band limits: 70091000
2019-12-30 09:24:45,858 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
2019-12-30 09:24:45,862 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 4950 | csdr fft_cc 4096 2867.383512544803 | csdr logaveragepower_cf -70 4096 93 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
closing client socket
Connection to 127.0.0.1 4950 port [tcp/*] succeeded!
client connection establised
2019-12-30 09:24:46,163 - owrx.websocket - ERROR - OSError while reading data; closing connection
Traceback (most recent call last):
  File "/home/agerik/openwebrx/owrx/websocket.py", line 202, in read_loop
    self.messageHandler.handleTextMessage(self, message)
  File "/home/agerik/openwebrx/owrx/connection.py", line 357, in handleTextMessage
    client = OpenWebRxReceiverClient(conn)
  File "/home/agerik/openwebrx/owrx/connection.py", line 113, in __init__
    features = FeatureDetector().feature_availability()
  File "/home/agerik/openwebrx/owrx/feature.py", line 38, in feature_availability
    return {name: self.is_available(name) for name in FeatureDetector.features}
  File "/home/agerik/openwebrx/owrx/feature.py", line 38, in <dictcomp>
    return {name: self.is_available(name) for name in FeatureDetector.features}
  File "/home/agerik/openwebrx/owrx/feature.py", line 60, in is_available
    return self.has_requirements(self.get_requirements(feature))
  File "/home/agerik/openwebrx/owrx/feature.py", line 71, in has_requirements
    passed = passed and self.has_requirement(requirement)
  File "/home/agerik/openwebrx/owrx/feature.py", line 83, in has_requirement
    return method()
  File "/home/agerik/openwebrx/owrx/feature.py", line 240, in has_soapy_sdrplay
    return self._has_soapy_driver("sdrPlay")
  File "/home/agerik/openwebrx/owrx/feature.py", line 224, in _has_soapy_driver
    process = subprocess.Popen(["SoapySDRUtil", "--info"], stdout=subprocess.PIPE, stderr=subprocess.DEVNULL)
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'SoapySDRUtil': 'SoapySDRUtil'
2019-12-30 09:24:46,181 - owrx.websocket - DEBUG - websocket loop ended; shutting down
2019-12-30 09:24:46,183 - owrx.websocket - DEBUG - websocket loop ended; sending close frame
Exception in thread Thread-68:
Traceback (most recent call last):
[... 121 lines cut ...]
Exception in thread Thread-83:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/agerik/openwebrx/csdr/csdr.py", line 59, in copy
    write(data)
  File "/home/agerik/openwebrx/owrx/connection.py", line 268, in write_s_meter_level
    self.send({"type": "smeter", "value": level})
  File "/home/agerik/openwebrx/owrx/connection.py", line 40, in send
    self.conn.send(data)
  File "/home/agerik/openwebrx/owrx/websocket.py", line 105, in send
    raise WebSocketClosed()
owrx.websocket.WebSocketClosed

^C^CException ignored in: <module 'threading' from '/usr/lib/python3.6/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

jketterl, 2019-12-30:
looks like the soapy sdr driver check needs additional protection for when SoapySDR is not installed...

```
FileNotFoundError: [Errno 2] No such file or directory: 'SoapySDRUtil': 'SoapySDRUtil'
```

I can fix that, just give me a minute.

interesting though that it worked before.

ofadam, 2019-12-30:
I think that I actually didn't have SoapySDR installed at first (I'm only
using an RTL-SDR with it at this point), but installed the packages to see
if it would fix a different problem I had. That may have triggered this!

jketterl, 2019-12-30:
yeah it should still work without. if you're running a checkout, just do a `git pull` now, the commit i did should fix it.

ofadam, 2019-12-30:
That did indeed fix the loop I was getting! Now I need to figure out why I'm getting an "usb_claim_interface error -6" when it tries to open the RTL-SDR...

jketterl, 2019-12-30:
probably a leftover from the previous errors. try to kill all instances of csdr and rtl_connector. that's something that happens a lot when openwebrx gets interrupted, and i can't seem to handle this properly due to the piping.

ofadam, 2019-12-30:
Killing those instances fixed it – thanks again for all your help. Now to
work on the antenna setup for this little remote netbook!

closed 2019-12-30 by jketterl

D0han, 2019-12-30:
> probably a leftover from the previous errors. try to kill all instances of csdr and rtl_connector. that's something that happens a lot when openwebrx gets interrupted, and i can't seem to handle this properly due to the piping.

Regarding that - consider using sessions or process groups to keep track of all spawned processes to kill them all when needed. Check https://docs.python.org/3/library/subprocess.html#popen-constructor and `start_new_session`.

jketterl, 2019-12-31:
it's alreay using process groups for the pipelines... the problem arises when a single process of the pipeline dies. this usually causes some unpredictable behaviour.