jketterl/openwebrx #45: Error during start

sm4xas, 2020-01-16:
Just cloned the rep, as I did a few days ago on a Pi.

Got this error:

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>

Traceback (most recent call last):
  File "./openwebrx.py", line 6, in <module>
    main()
  File "/home/pi/openwebrx/owrx/__main__.py", line 33, in main
    pm = PropertyManager.getSharedInstance().loadConfig()
  File "/home/pi/openwebrx/owrx/config.py", line 140, in loadConfig
    cfg = importlib.util.module_from_spec(spec)
AttributeError: 'module' object has no attribute 'module_from_spec'

Any ideas?

jketterl, 2020-01-16:
could be the python version. what are you using? minimum is 3.5 right now.

jketterl, 2020-01-16:
the docker version doesn't include the rtl_sdr soapy driver. i can include it in the next build, if that helps.

sm4xas, 2020-01-16:
Please do. I'll try it on my other Pi3, right now I'm testing on my Pi2 and it's probably too old dist. The latest python in the reps is 3.4.2.

jketterl, 2020-01-17:
alright, the builds just went through, docker images should now include the soapy module. the config still needs to be edited though.

jketterl, 2020-03-11:
closing this since `rtl_sdr_soapy` and `direct_sampling` are reported working in multiple cases.

closed 2020-03-11 by jketterl