jketterl/openwebrx #46: Feature request: possibility to connect to CW Skimmer

dh5ym, 2020-01-17:
Using an openwebrx as sample source for a CW Skimmer would be a nice to have.
CW Skimmer gets its samples via audio interface (max. 192kSpl/s IQ) or via network interface (i never used it). A background IQ service on the openwebrx server could deliver samples via loopback audio (possibly network is easier?). Furthermore the center frequency of the skimmer needs to be set via a telnet connection (or somehow read by skimmer via a CAT control).
BTW: there is also an open source bayesian statistics CW decoder by AG1LE but unfortunately it it has no CW signal detection that could start the decoder on each signal it finds in a given spectrum. Make a cwskimmer like application would certainly be a separate project.

jketterl, 2020-01-30:
out of curiosity, i just did a quick search on google... turns out cwskimmer is a windows only application, which is a major showstopper.

any idea about alternatives?

emulation via wine or similiar is out of question. it's too much of a mess and won't work properly across all platforms.

dh5ym, 2020-01-31:
Correct. CWSkimmer is a Win Application. It works with Wine, just needs quite some CPU. I use this configuration for quite some while now with Quisk.
An option would be to utilize the work of AG1LE. He already wrote an Bayesian statistics CW decoder. But for Skimmer-like functionality there is quite some work missing. From the samples some classification needs to be done in order to identify the CW signals. Something like that was already in some older fldigi code (which was using the decoder from Mauri). Then for each of those signals the decoder needs to be started, callsigns filtered out etc. Its certainly some extra project and nothing that can be done on as a side activity.

dh5ym, 2020-01-31:
This was the last entry on the Bayesian decoder on the blog of AG1LE:
http://ag1le.blogspot.com/2014/07/new-morse-decoder-part-6.html
His github repository on the topic is here:
https://github.com/ag1le/morse-wip
Maybe he has more recent stuff elsewhere.

Nammurg, 2021-09-25:
Maybe RSCW can do the trick?
https://www.pa3fwm.nl/software/rscw/

scannerbrightly, 2023-11-12:
This is a great idea, and it looks like Multimon-ng that Nammurg suggested above is very flexible. It also decodes others things as well, which might be useful.