jketterl/openwebrx #47: Changelog for new features

ofadam, 2020-01-17:
I noticed that the readme contains a changelog that was last updated in November – have all updates since then been bug fixes, or have there been new features rolled out? Any plans for creating a separate changelog?

jketterl, 2020-01-17:
the changelog has been updated whenever i rolled out a raspberry pi image so far. i've been working a lot on build automation, which has made the whole project shift more to a "continuous" development process. i'm currently trying to find some kind of a "sweet spot" where i could tag the next major release.

also, with the changelog filling up, i have been noticing that the readme is getting cluttered and hard to use. so i was planning to move it to another location, probably some changelog.md.

ofadam, 2020-01-17:
Perfect! Thanks again for all your work on keeping this alive – I'm still
amazed at what's possible with radios now compared to even just a short
time ago.

commit b35958c6 (2020-01-29): update changelog, closes #47
closed 2020-01-29 by jketterl