jketterl/openwebrx #49: Feature Request - gain control on GUI

mripv6, 2020-01-24:
I would like to see a gain control in the Web GUI so I can adjust the gain of the SDRPlay on the fly. 

I know I should open another issue, but this is somewhat related. I would also like to see the max/min waterfall controls automatically adjusted to provide a reasonable display. When I switch bands, I have to change the waterfall sliders. The sliders are very touchy, too.

Thanks for keeping the project alive.

jketterl, 2020-01-25:
this is one of the more refined features that will have to wait. right now it's up to the receiver operator to set up a sensible gain per profile, which works for most scenarios.

allowing gain control on the interface will have a huge impact when the receiver is used by multiple users, it is one of the parameters that will ruin the experience for everybody if one connected user is e.g. inexperienced or malicious. this may find its place in a full-control manual mode, but it should be up to the receiver operator to enable it.

as for the waterfall: the default waterfall levels can be set in the config for each profile using the `waterfall_max_level` and `waterfall_min_level` keys. It's a little complicated at the moment since the actual values need to be obtained from the developer tool of your browsers, but once set up it will make life much easier.

there's also the "Auto-adjust waterfall colors" button, on the left end of the waterfall minimum slider. it's usually sufficient on rtl-sdr devices... not so much on the sdrplay, since it gets the minimum wrong due to it's narrower input filter. will need some work.

also, have you tried to use the mousewheel on the sliders? in my opinion, this gives better control over the fine adjustments, and also gives immediate feedback, as opposed to drag and drop.

mripv6, 2020-01-25:
Thank you for your response. I found the auto-adjust, but it wasn't providing a color scheme with sufficient dynamic range, which is why I was adjusting the sliders.  I'll search for the waterfall_max_level with the browser dev tools and give that a try. 

Thanks for the tip on the sliders. Using the mouse while makes it much easier to fine tune the display!

I see your point about the gain control. I suspect you're correct. After I figure out the right gain setting, it probably won't change too much. So a static value in the profile is probably the way to go.

closed 2020-01-25 by mripv6