jketterl/openwebrx #50: Crash with Sdrplay

LanWolf, 2020-01-28:
I tried using the image (jan 2020)  with an sdrplay rsp1 , but is keeps crashing/locking up. sometimes it runs a few minutes, but when changing anything it locks up totally.
Maybe a driver problem ?

Attached an Airspy HF + Discovery, and it runs fine for days now 
(except the gain setting, but that already is a known issue afiak)

jketterl, 2020-01-29:
difficult to tell. can you please check if there's any messages related to new usb devices / usb devices being disconnected in `sudo dmesg -T` when it happens? i'm suspecting its the hardware, and my suggestion would be to improve the power supply and / or use a powered usb hub.

the driver itself... well, it has its issues, but those are mostly related to initial startup. once it running, it's usually fine, so that would be a new one.

also: i thought i had the airspy gain issue sorted by compiling soapy and the airspyhf module from source. i need to double check that.

LanWolf, 2020-02-03:
I cannot check messages during crash, beacuse it locks up the whole pi. have to powercycle to get going again. 
It mostly hapens when closing the webpage, sliding the waterfall sliders.

What would be the correct gain setting for the airspy ? it was 20, i tried 30 even but no succes

jketterl, 2020-02-03:
if it completely locks up the system it's most likely a hardware or driver fault, you'd probably get the same result with other sdr software. you can try the suggestions above, but since you say it happens during disconnect: try setting `"always-on": True` on the sdrplay device in your config. That's a workaround for sdr devices that cannot reliably shut down / start up at will. Basically it will keep the process that interfaces with your sdr running in the background, avoiding the shutdown.

as for the airspy: you're right, gain control does not work as expected. when i though i had sorted it out, all i got was the correct listing of gain values from `SoapySDRUtil --probe`, but setting them doesn't work. I will need to investigate this...

jketterl, 2020-02-15:
branching the airspy gain issue into #53

closed 2020-03-21 by jketterl