jketterl/openwebrx #52: wget on docket with error

MidnightRAT, 2020-02-14:
2020-02-14 21:02:57,377 - owrx.sdrhu - DEBUG - wget --timeout=15 -4qO- https://sdr.hu/update --post-data "url=http://websdr.domain.pp.ua:8073&apikey=qwertyuiasdfghjkzxcvbnm" 2>&1
2020-02-14 21:02:57,383 - owrx.sdrhu - WARNING - wget failed while updating, your receiver cannot be listed on sdr.hu!

If i change parametrs to
wget -qO- https://sdr.hu/update --post-data "url=http://websdr.domain.pp.ua:8073&apikey=qwertyuiasdfghjkzxcvbnm" 2>&1
Its work.

jketterl, 2020-02-14:
thanks for letting me know, i'll need to look into this. it works outside of docker, so i'm suspecting that the wget in alpine has been modified or may not even be an original wget. guess the easiest fix will be to replace wget with python-native http methods, there's no real need have this run externally.

jketterl, 2020-02-14:
confirmed: wget in alpine has been replaced with busybox, so simply does not support the full feature set.

jketterl, 2020-02-15:
would you mind testing out the latest docker build? it has a fix, but i'm unable to test myself since i'm on an ipv6 connection and cannot set up port forwarding on the ipv4 tunnel.

MidnightRAT, 2020-02-16:
Sorry for my English
1. Not work URL: https://physics.princeton.edu/pulsar/k1jt/wsjtx.html (May be it https://github.com/roelandjansen/wsjt-x)
2. I could not your version of openwebrh find device until I launched the original version.
I saw in dmsg
[1600.029814] r820t 2-001a: destroying instance
[1600.029961] dvb_usb_v2: 'Realtek RTL2832U reference design: 3-4' successfully deinitialized and disconnected
After that, your assembly started.

MidnightRAT, 2020-02-16:
Its work on 
http://sdr.chirik.pp.ua:8073/#freq=438900000,mod=dmr
DMR not start past URL. I need change FM, then DMR.

You can add sshd to container
And i can try forward ssh port to container.
And you can debug you project

jketterl, 2020-02-17:
would you mind how you ended up on the wsjt-x homepage? i'm a little confused. i hope you didn't interpret my message as in you would need to build it yourself. The change should be in the latest docker build (i.e. images on [https://hub.docker.com/u/jketterl](https://hub.docker.com/u/jketterl)) - you just need to `docker pull` the latest update from there.

I can see your receiver is now listed on sdr.hu, so I suppose at least something worked?

jketterl, 2020-03-11:
any update on this? not sure if it works by now, but don't have any actions to take here at this point.

closed 2020-03-21 by jketterl