jketterl/openwebrx #54: Crash if there are multiple clients using Packet decoder

dl9rdz, 2020-02-16:
On an Odroid XU4 with just a simple output device, starting direwolf twice for two openwebrx clients will result in a failure as the default audio output device cannot be opened twice. There is a simple fix to this: Just tell direwolf to use the "null" output device. 

(Sorry that this is not a nice pull request, but its just one line to change anyway...)

diff --git a/owrx/kiss.py b/owrx/kiss.py
index 1ea9408..d47fa4d 100644
--- a/owrx/kiss.py
+++ b/owrx/kiss.py
@@ -18,6 +18,7 @@ class DirewolfConfig(object):
 
         config = """
 ACHANNELS 1
+ADEVICE stdin null
 
 CHANNEL 0
 MYCALL {callsign}

jketterl, 2020-02-17:
oh that's very good. I was searching all over, how i could disable the audio output of direwolf, never thought about redirecting it to nowhere.

jketterl, 2020-02-17:
included in ad565c5a2b654b5728e53af0aae1bcc57d781d1c

closed 2020-02-27 by jketterl