jketterl/openwebrx #55: No audio, no waterfall, no error messages

ofadam, 2020-02-18:
I'm having some difficulty getting it to run under Ubuntu 18 - I installed all dependencies and configured openwebrx correctly for an RSP2Pro... but when I launch the browser interface, it seems like there's no signal information actually being capture (no audio, no waterfall.) I even checked the /features page and everything looks good. 

Any advice on how to troubleshoot? I don't see anything in the console log that looks amiss, either.

jketterl, 2020-02-18:
re-plug the RSP... sounds stupid, but i've had that quite often. they don't work every time. not sure if it's software or hardware related.

ofadam, 2020-02-18:
I tried that, but without success. I'm able to see the RSP2pro via SoapySDRUtil.

jketterl, 2020-02-18:
can you get anything out of it with some other sdr software?

ofadam, 2020-02-18:
SoapySDRUtil export: 
`######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=sdrplay
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP2
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=2
  serial=1708071110

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
`

ofadam, 2020-02-18:
The console from openwebrx: 

`2020-02-18 19:40:24,307 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: SDRPlay RSP2
2020-02-18 19:40:38,798 - owrx.connection - DEBUG - client connection intitialized
2020-02-18 19:40:38,803 - owrx.source - DEBUG - activating profile 20m
2020-02-18 19:40:38,805 - owrx.source - INFO - Started sdr source: soapy_connector -s 500000 -P 0 -g 0 -d driver=sdrplay -a "Antenna B" -f 14150000 -p 49407 -c 35727
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Tuner gain set to 0.00 dB.
IQ worker thread started
[INFO] Using format CS16.
socket setup complete, waiting for connections
setting up control socket...
control socket started on 35727
client connection establised
2020-02-18 19:40:40,584 - owrx.source.connector - DEBUG - opening control socket...
control connection established
2020-02-18 19:40:40,584 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
2020-02-18 19:40:40,586 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 49407 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_139993258928728_shift_pipe | csdr fir_decimate_cc 45 0.0033333333333333335 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_139993258928728_bpf_pipe 0.0288 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_139993258928728_squelch_pipe --outfifo /tmp/openwebrx_pipe_139993258928728_smeter_pipe 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.0078105316200554 | csdr deemphasis_nfm_ff 11025 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
2020-02-18 19:40:40,593 - owrx.dsp - DEBUG - adding new output of type audio
csdr bandpass_fir_fft_cc: fifo control mode on
fir_decimate_cc: taps_length = 1199
csdr squelch_and_smeter_cc: fifo control mode on
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr shift_addition_cc: fifo control mode on
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
2020-02-18 19:40:40,600 - owrx.dsp - DEBUG - adding new output of type smeter
client connection establised
Connection to 127.0.0.1 49407 port [tcp/*] succeeded!
2020-02-18 19:40:40,604 - owrx.bands - WARNING - Frequency for wspr on 4m is not within band limits: 70091000
2020-02-18 19:40:40,608 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
(overlap_length = 138) = taps_length - 1
csdr shift_addition_cc: reinitialized to -0
2020-02-18 19:40:40,611 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 49407 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
Connection to 127.0.0.1 49407 port [tcp/*] succeeded!
client connection establised
csdr bandpass_fir_fft_cc: filter initialized, low_cut = -0.36, high_cut = 0.36
2020-02-18 19:40:41,226 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
2020-02-18 19:40:41,229 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 49407 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_139993258928728_shift_pipe | csdr fir_decimate_cc 41 0.0036585365853658534 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_139993258928728_bpf_pipe 0.02624 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_139993258928728_squelch_pipe --outfifo /tmp/openwebrx_pipe_139993258928728_smeter_pipe 5 2 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.016260162601626 | csdr deemphasis_nfm_ff 12000 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
2020-02-18 19:40:41,234 - owrx.dsp - DEBUG - adding new output of type audio
Connection to 127.0.0.1 49407 port [tcp/*] succeeded!
csdr shift_addition_cc: fifo control mode on
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr squelch_and_smeter_cc: fifo control mode on
client connection establised
csdr bandpass_fir_fft_cc: fifo control mode on
fir_decimate_cc: taps_length = 1093
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 153) + (input_size = 360) - 1
(overlap_length = 152) = taps_length - 1
csdr shift_addition_cc: reinitialized to -0
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
2020-02-18 19:40:41,253 - owrx.dsp - DEBUG - adding new output of type smeter
2020-02-18 19:40:41,255 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
2020-02-18 19:40:41,256 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 49407 | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_139993258928728_shift_pipe | csdr fir_decimate_cc 41 0.0036585365853658534 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_139993258928728_bpf_pipe 0.02624 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_139993258928728_squelch_pipe --outfifo /tmp/openwebrx_pipe_139993258928728_smeter_pipe 5 2 | csdr realpart_cf | csdr fractional_decimator_ff 1.016260162601626 | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
2020-02-18 19:40:41,258 - owrx.dsp - DEBUG - adding new output of type audio
csdr shift_addition_cc: fifo control mode on
csdr squelch_and_smeter_cc: fifo control mode on
Connection to 127.0.0.1 49407 port [tcp/*] succeeded!
fir_decimate_cc: taps_length = 1093
client connection establised
csdr bandpass_fir_fft_cc: fifo control mode on
[... cut ...] um_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 153) + (input_size = 360) - 1
(overlap_length = 152) = taps_length - 1
csdr shift_addition_cc: reinitialized to -0
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
2020-02-18 19:40:41,271 - owrx.dsp - DEBUG - adding new output of type smeter
csdr bandpass_fir_fft_cc: filter initialized, low_cut = 0.0246, high_cut = 0.246

`

jketterl, 2020-02-18:
SoapySDRUtil doesn't help, it will say the device is available, but it won't indicate if there's any data from it. And the console output... yes, there's no errors. can you try another SDR?

ofadam, 2020-02-18:
I’ll give that a try soon... I need to find the RTL-SDR stick I have.

ofadam, 2020-02-21:
Hmm, it appears something is wrong with my setup and the SDRPlay – I'll go ahead and close this issue.

closed 2020-02-21 by ofadam