jketterl/openwebrx #59: connection to plutosdr / soapy remote

tmlll, 2020-02-25:
Is it possible to connect openwebrx to plutosdr with SoapyRemote?

I have installed openwebrx on raspberry pi and I would like to connect it with plutosdr in SoapyRemote mode via tcp.

What do you need to enter in /etc/openwebrx/config_webrx.py in the sdrs = {} section?

jketterl, 2020-02-25:
It should be possible, even though that specific scenario hasn't been used yet, at least to my knowledge.

The `soapy_remote` sdr type follows pretty much the same config scheme as the other drivers apart from a few special keys: `remote` and `remote_driver`. So a very basic config example woud look like this:

```
sdrs = {
    "pluto": {
        "name": "Adalm Pluto",
        "type": "soapy_remote",
        "ppm": 0,
        "remote": "192.168.123.456",
        "remote_driver": "plutosdr",
        "profiles": {
            "20m": {
                "name": "20m",
                "center_freq": 14150000,
                "rf_gain": 20,
                "samp_rate": 500000,
                "start_freq": 14070000,
                "start_mod": "usb",
                "antenna": "Tuner 2 50 ohm",
            },
        },
    },
}
```

I'm not sure if the profile is even useful with a Pluto, I have only quickly adopted another config I had at hand.

As you can see, `remote` is the target IP or hostname, whereas `remote_driver` is the soapy driver identifier on the destination machine. `remote_driver` is optional, if omitted, the destination will choose the first available soapy device.

tmlll, 2020-02-27:
Thanks for your reply.

The example configuration has worked with PLUTOSDR since the first start.
OPENWEBRX worked correctly, I mainly tested how APRS decoding works.

The problem is that when after a while I turned ON the computer again and I opened the OPENWEBRX website, I could not connect to PLUTOSDR again.
I spent a long time trying to solve the problem but couldn't connect. 
Raspberry and PLUTOSDR restarts did not help. Finally I gave up.

The next day I came back to this.
After long tests I came to the conclusion that in order to run OPENWEBRX and PLUTOSDR I need to change the IP address with which SoapySDRServer starts.

For example, if it was before:

```
# SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-g03b483ad
Server UUID: bb724abf-0169-1000-8567-13e3007f0101
Launching the server... tcp://0.0.0.0:55132
Server bound to 0.0.0.0:55132
Launching discovery server...
Connecting to DNS-SD daemon...
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: pluto
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     pluto.local
[INFO] avahi_entry_group_add_service(pluto._soapy._tcp)
Press Ctrl+C to stop the server
```
next time I launched:

```
# SoapySDRServer --bind="192.168.20.1:55132"
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-g03b483ad
Server UUID: d6ac5e4e-0015-1000-8567-0512007f0101
Launching the server... tcp://192.168.20.1:55132
Server bound to 192.168.20.1:55132
Launching discovery server...
Connecting to DNS-SD daemon...
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: pluto
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     pluto.local
[INFO] avahi_entry_group_add_service(pluto._soapy._tcp)
Press Ctrl+C to stop the server
```
When **OPENWEBRX is working properly**, SoapySDRServer only establishes the following connections:

```
SoapyServerListener::accept(192.168.9.7:44746)
SoapyServerListener::accept(192.168.9.7:44764)
SoapyServerListener::accept(192.168.9.7:44768)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.9.7:44770)
```

During normal operation, the OPENWEBRX log looks like this:

```
-- Logs begin at Thu 2020-02-27 16:29:24 UTC, end at Thu 2020-02-27 16:32:47 UTC. --
Feb 27 16:31:59 openwebrx systemd[1]: Started OpenWebRX WebSDR receiver.
[... 197 lines cut ...]
```

The above logs show the connection of only one OPENWEBRX client.
I tried different connections PLUTOSDR, LAN, WLAN, different IP. All on the local network.
The connection can only be established after reboot raspberry and pluto and when the SoapySDRServer address changes.

I use in openwebrx config:

`"remote": "192.168.20.1"
`

I also use this PLUTOSDR with Soapy Remote and  `SDR Console` and I have no problems with it.
I have no more ideas. I use experimental repo for raspberry now, but the problem is the same.

I think this is a `soapy_connector` problem, but maybe I'm wrong.

After issuing the following command, SoapySDRServer behaves similarly, closes the connection repeatedly, and reconnects:

`
soapy_connector -s 2400000 -P 0 -g 30 -d driver=remote,remote=192.168.20.1,remote:driver=plutosdr -f 438800000 -p 38181 -c 37237
`

I also did short tests on another machine, running OPENWEBRX in a container but without success.

If you had a problem with it or maybe you had an idea, please help.

jketterl, 2020-02-27:
well, i don't have any immediate conclusions. the soapy_connector doesn't do much with the information you give it, it mostly just passes them to the soapy libraries. I don't see any errors either, which makes debugging harder.

I see you are running the latest SoapySDRServer version, but I don't see which client library version you have installed. If you did install from the package repository, you're probably running version 0.4.3, which is a bit dated. It might be worth trying the current version (https://github.com/pothosware/SoapyRemote) on the client, but you'll need to install it from source.

Another thing that might help (at least as a work-around) is the `"always-on": True` option, which will keep the connection open and running. Of course that will cause network traffic at all time in your case, but it should prevent this issue.

Heppie, 2020-02-27:
Sorry to Hijack this issue but I dont think it is its worth its own issue.

I am trying to use soapyremote to connect to an RTLSDR but I get:

owrx.sdr - ERROR - The RTL source type "soapy_remote" is not available. please check requirements.

Have I missed something in the SDR config?

`sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR",
        "type": "soapy_remote",
        "ppm": 0,
        "remote": "192.168.0.56",
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 439275000,
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },
}`

jketterl, 2020-02-27:
@Heppie you're probably missing a piece of software. Please check the feature report by appending "/features" to your receiver URL (e.g. http://localhost:8073/features) and make sure that all dependencies listed under "soapy_remote" are available. Depending on what installation method you went with, you'll need some extra packages or need to compile from source.

closed 2020-03-21 by jketterl