jketterl/openwebrx #62: WSPR upload to wsprnet.org (feature)

azwirko, 2020-02-29:
Reading the Background Decoding docs 

https://github.com/jketterl/openwebrx/wiki/Background-decoding

I noted you cannot currently upload to wsprnet.org.  I was wondering if the work already done for the KiwiSDR would be useful for providing an upload capability  See wsprdaemon:

https://github.com/rrobinett/wsprdaemon

It uses kiwiclient/kiwirecorder to connect to a KiwiSDR and produces WAV files which are then handed off to the decoder and then uploaded.  If you could produce similar formatted WAV files I would think the rest of the app could be utilized to provide the upload capability to wsprnet.org

73 

andyz - K1RA

jketterl, 2020-02-29:
Thanks for the suggestion, but decoding WSPR is already implemented (pretty much in the way you describe). I cannot report to wsprnet since their API is lacking in a number of ways.

First, it's completely undocumented and you have to reverse engineer what WSJT-X does.
Second, what I have found is basically summed up as "upload the `ALL_WSPR.TXT` file". And, as simple as that may sound, the decoder does not write that file in a safe manner, in particular: It is not handling it well enough to be used in a multi-threaded environment.
Third, OpenWebRX is parsing the output of the decoder STDOUT. That gives sufficient information for most cases, but there's some fields missing, and there's no way of getting them (I even went so far as to checkout the WSJT-X source code to make sure).

So, as of right now, reporting to wsprnet has been put to the back of the feature list, since it will consume big amounts of time to come up with a solution on my end. It would be much easier for wsprnet to come up with a sensible API.

I'm labelling this as a feature request. It has been on my personal list for quite some time, but with these findings I had to put this aside for now.

jketterl, 2021-01-15:
Alright... I did some reverse engineering (I am slowly getting an grip on the WSJT-X sourcecode) and some backtracing and used the information i gathered that way to implement WSPRnet spot uploads. It seems very crude to me (spots are not batched in any way, every spot is it's own HTTP request) and since I still don't have any documentation to work off, I cannot really ensure that it will work tomorrow. It seems to work today.

The WSPRnet spotting endpoint seems to be a bit unreliable, I have already seen plenty of connection resets or timeouts so far. This is beyond my control, of course, but if this is a regular thing, I may need to silence these errors in the log.

closed 2021-02-01 by jketterl