jketterl/openwebrx #63: csdr 0.15.1 has some problems

kolbma, 2020-03-03:
https://github.com/jketterl/csdr 0.15.1

Try to compile it with -Wall

There are some functions without returns but with return types.
And uninitialized stuff.

Packaging of this code fails.

jketterl, 2020-03-05:
I know that there's quite a few problems with it. But as you may have noticed, there's no issues on that project: That's because I'm only maintaining that fork for the necessary fixes to keep OpenWebRX running. It is officially still maintained upstream, and I have pull requests open myself for different things.

I think you should address these issues here: https://github.com/ha7ilm/csdr

I may occassionally get around to clean this up properly, but right now, it'll have to do the way it is.

Packaging does not fail on my end, though, not sure what that's happening there.

VVelox, 2020-03-10:
@jketterl Sadly ha7ilm is really no longer maintaining csdr as well.

What is really needed to is some one needs to fork it and start properly accepting patches. My fix to de-fuck building on non-Linux OS has been stagnating for a month now.

jketterl, 2020-03-11:
I'm very well aware of the situation. As I said before, I do have open pull requests myself. In his latest announcement however, András has clearly stated that the "[...] CSDR project will still be maintained", so this may be temporary. The lack of feedback is no immediate reason for me to fork csdr.

I'm closing this issue for now since i don't think we're going anywhere with this. The only reason I kept this open was because I was curious about what packaging the original author was performing. Feel free to follow up if there's any information regarding that.

closed 2020-03-11 by jketterl