jketterl/openwebrx #65: Squelch doesn't work in DIGI Packet

kolbma, 2020-03-03:
At least in DIGI **Packet** mode the squelch setting doesn't do anything.
Maybe the reason is that the SQU is not fast enough for digi mode.
But then a feature for "audio output squelch" would be fine, to simply mute the noise below db level.

jketterl, 2020-03-05:
That is pretty much intended. The digital decoders don't mind the noise, the squelch can only have negative effects in that case.

Unfortunately, the way things are setup, I cannot have a separate squelch on the audio since that would mean I'd have to run a second demodulator, which I'm not going to do as the additional CPU usage is unacceptable.

This also affects Pocsag.

kolbma, 2020-03-11:
Ok. Then what do you think about disabling the squelch control in these modes?

jketterl, 2020-03-11:
definitely the way to go, but need to have a look if it's realistic with the current codebase (it's a lot of spaghetti code). i may need to rewrite some more stuff first.

jketterl, 2020-05-21:
looks like I only referenced this in the commits, but didn't post an update. There's been a rework of a lot of the frontend code, and I included the suggestion in that move, so the squelch control is now disabled on the modes that don't support it.

closed 2020-05-21 by jketterl