jketterl/openwebrx #66: sdrplay fm/dab notches

ra1nb0w, 2020-03-03:
first of all thanks for maintining alive the project!
one feature that I need to support broadcast radios is to enable/disable the fm(mw/lf) and dab notches of the sdrplay devices. With soapy I generaly set `rfnotch_ctrl=true,dabnotch_ctrlb=true` as settings based on which band I want to listen. Should be nice to have these configs into `profiles`. Another useful conf is `biasT_ctrl`.

thanks

jketterl, 2020-03-05:
bias tee has been requested on the mailing list already, the others shouldn't be a lot of extra work :)

ra1nb0w, 2020-03-05:
thank you to take it in consideration :)
I will be happy to test the changes and to confirm that they work fine.

jketterl, 2020-03-14:
alright, i've added a new settings mapper and included the mappings in https://github.com/jketterl/openwebrx/commit/fea2cd1cc5c00ac2fb7d4726cb72f51235e586ee

please note that the settings are called `rf_notch`, `dab_notch` and `bias_tee` respectively, to keep them in the general naming scheme.

I haven't tested it yet myself, so this may fail completely.

It may be necessary to use literal strings, i.e. `"rf_notch": "true"` depending on whether or not soapy interprets booleans in a case-sensitive manner.

ra1nb0w, 2020-03-14:
updated owrx and owrx_connector but changing from `"rf_notch": "true"` (default) to `"rf_notch": "false"` seems doesn't change the spectrum.
Do I need to update something else?
How can I debug?

thank you

jketterl, 2020-03-14:
In the OpenWebRX log/output: See what paramters are passed to soapy_connector, you should see the `-t` option with the corresponding `rfnotch_ctrl` and `dabnotch_ctrl` values.

When switching profiles, watch out for these messages: `sending property change over control socket: settings changed to {1}`

Did you pull the owrx_connector from the develop branch? There has been a switch from master to develop with the 0.18.0 release.

jketterl, 2020-03-14:
I did some testing myself today, and it seems to work so far. Here's what it looks like if i turn `rf_notch` on and off right in the middle of the FM broadcast band:

[image]

I must say, I'm pretty impressed with the filter of the SDRplay, there's nothing left!

jketterl, 2020-03-14:
i can also confirm that it's indeed necessary to use strings. so basically, the settings must look like this:

```
"rf_notch": "false",
"dab_notch": "false",
```

python's boolean values `True` / `False` will not work.

jketterl, 2020-03-14:
I just included a custom conversion for the latter since it certainly would become an issue at some point, soooo.... `git pull` and disregard.

ra1nb0w, 2020-03-15:
sorry, I was to much concise.
Yes `"false"` was need; thanks for the change.
ok, now I have understood why I didn't see the `soapy_connector -t`. I was using `*_notch` only at the level profile; defining it also outside I get the minus t; therefore must be declared outside to work also in the profile. I can sent different values per profile.
(before this issue I implemented the same changes and it didn't work for the same reason)
Thanks!

Just a note: the signal is lower than CubicSDR or SDRuno for the same band; see below

CubicSDR
[image]

OpenWebRX
[image]

same _rf_gain=2_ on both program and both are using the same soapysdrsdrplay version.
Do you have some ideas?

jketterl, 2020-03-15:
I just tried the profile level, it works pretty much as expected (ie. it shows the same erratic behaviour as other settings when omitted in some profiles). General rule right now is: if you want to use any setting on profile level, you'll need to include it in all profiles. That's because of how the profiles work internally, they don't have a mechanism to fall back to defaults when keys are missing.

The different visual representation may be down to the watefall level adjustments. It seems like your openwebrx screenshot doesn't use the available range of the color profile (ie. no yellow or blue), which probably gives the impression of lower levels. At least that's what I'd interpret with what I can see, you can try setting a lower max level to confirm.

ra1nb0w, 2020-03-16:
> I just tried the profile level, it works pretty much as expected (ie. it shows the same erratic behaviour as other settings when omitted in some profiles). General rule right now is: if you want to [...]

ok; probably a note in the config can help to avoid mistake.

> The different visual representation may be down to the watefall level adjustments. It seems like your openwebrx screenshot doesn't use the available range of the color profile (ie. no yellow or [...]

I do more tests. thank you!

ra1nb0w, 2020-03-16:
feature implemented

closed 2020-03-16 by ra1nb0w