jketterl/openwebrx #67: Info ONLY: Workarounds for openwebrx CPU Load Reduce + Hanging IO

linuxonlinehelp, 2020-03-05:
[https://github.com/linuxonlinehelp/openwebrx-workarounds-bugfixes/wiki/Openwebrx-Help-Workarounds](url)

hope its useful..

D0han, 2020-03-05:
This link is not correct, can you fix markdown for it?

jketterl, 2020-03-05:
Even with the correct link, it's not very helpful since most of the instructions on that page are based on the "original" version and are no longer relevant.

- Passing the Squelch as a URL fragment is no longer supported. Instead, you can set the `initial_squelch_level` on the config profiles.
- The startup delay of the FFT functions has been fixed and is no longer an issue when you're using my fork of csdr (I also sent the change upstream, however there's no attention whatsoever... here's the pull request: https://github.com/ha7ilm/csdr/pull/51)
- there's no index.wrx any more.

Please stop bringing in outdated information, it is only confusing for users.

closed 2020-03-05 by jketterl

linuxonlinehelp, 2020-03-06:
i purged it! iam out, (the old developer is a friend of me)