jketterl/openwebrx #68: Initial setup for Docker. Probably a silly question but wanted to ask.

thetechlink, 2020-03-09:
I have installed and configured openwebrx in the past from the origin. Got it working and there was significant setup for RTLSDR or RSPDUO before you setup the web server to get openwebrx  working.  You had to install the following dependencies
    libcsdr
    rtl-sdr

Does the docker image take care all of the dependencies like libcsdr, rtl-sdr and any other  drivers for SDR's like RSPDUO or do you still have to do the same work as before to setup SDR driver on the docker host? 

With the new fork I have gotten docker to work on a Raspberry 4 that was configured for the previous upsteam provider with RTLSDR but could not get the SDRPlay to work. 

Now I am installing on a fresh build of Ubuntu on a more powerful 18.04.4 x64 box and am having a hard time getting anything to work Am i missing something in my understanding? Is there a different way to configure drivers than the upstream project use to? 

I am using the following command to launch the docker container

sudo docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-config:/etc/openwebrx jketterl/openwebrx --tmpfs=/tmp/openwebrx

This is the error I get when trying to log in for the first time. 
This receiver is currently unavailable due to technical issues.
Error Message:
No SDR Devices available

Do I need to specify the specific usb device?

Thanks in advance for your time,
Zac

jketterl, 2020-03-09:
Hello Zac,

the docker image should indeed take care of all the driver stuff, it does support most of the available SDR options currently supported by OpenWebRX. So the basic idea is to just run it and enjoy it.

Unfortunately, you are probably hitting this old issue https://github.com/jketterl/openwebrx/issues/7 (I'm not sure why it's closed. it hasn't been resolved as far as i know...)

Quick summary: The driver for SDRplay devices unfortunately is linked with glibc, whereas the docker images use Alpine and musl, which is missing some symbols, which in turn makes anything that tries to use the driver crash, including the soapy_connector used by OpenWebRX.

I have been in touch with SDRplay, and they have directed me towards the newer version 3 of the driver, which is a major upgrade, probably involving some more work. I have since not been able to address the issue: https://github.com/jketterl/openwebrx/issues/36

So, as of right now, you'll have to chose another option for the SDRplay, at least until this issue is resolved.

Sorry about that.
Jakob

thetechlink, 2020-03-09:
So I installed another fresh copy of ubuntu on another laptop. I installed docker and ran docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-config:/etc/openwebrx jketterl/openwebrx --tmpfs=/tmp/openwebrx

I have my nesdr smart by nooelec.com SDR plugged in and is shows in lsusb as Bus 001 Device 005: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

When I launch openwebrx via localhost no receivers are found. 
[https://imgur.com/6yaRTI0](url)

This is working on my raspberry pi 4 but again I had followed the setup guide on it for the older version of openwebrx. Do I need to blacklist the rtlsdr so it does not try to install a system driver?

I am sure I am missing something basic or simple here but its just not working. 

Again, I really do appreciate your time and the huge improvements you have made. 
Zac

jketterl, 2020-03-09:
Did you plug in the rtlsdr stick _before_ bringing up the docker container? That's important because docker does not sync the /dev/bus/usb filesystem after container startup.

jketterl, 2020-03-09:
Actually, after figuring out how to open the screenshot: yes, it looks like something is blocking the USB device. Can you check for the `dvb_usb_rtl28xxu` module on the docker host?

thetechlink, 2020-03-10:
Ok, got it figured out. 

I followed this from https://blog.sdr.hu/2015/06/30/quick-setup-openwebrx.html

#Disable the DVB-T driver, which would prevent the rtl_sdr tool from accessing the stick
#(if you want to use it for DVB-T reception later, you should undo this change):
sudo bash -c 'echo -e "\n# for RTL-SDR:\nblacklist dvb_usb_rtl28xxu\n" >> /etc/modprobe.d/blacklist.conf'
#The following line is only needed on Ubuntu 16.04 or newer to apply changes made to the blacklist settings.
sudo update-initramfs -u #only on Ubuntu 16.04 or newer
#Disable the problematic kernel module for the current session:
sudo rmmod dvb_usb_rtl28xxu 

Now its all good.

I will do a setup video on youtube for the rtlsdr config and the setup for my limesdr sdr as well

It would be great if that was noted in the setup wiki. I don't mind helping with documentation if you need it. I am no Linux rocker engineer but I could help document configs if you need help. 

Thanks,
Zac

closed 2020-03-10 by thetechlink

jketterl, 2020-03-11:
I have includeded a corresponding paragraph in the Wiki: https://github.com/jketterl/openwebrx/wiki/Getting-Started-using-Docker#known-issues