jketterl/openwebrx #69: Gain settings for SDRPlay

ofadam, 2020-03-09:
The default configuration for SDRPlay seems to have the gain set to "0" - is that autogain? What's the range that parameter can be set to?

jketterl, 2020-03-09:
The different models of SDRplay devices have different architectures and gain values, so there's not real answer to that question.

The value is pretty much passed straight to SoapySDR, so there's not much to report as far as OpenWebRX goes. You can also set the individual gain levels by passing a string like `"RFGR=1"`. You can check out the available settings using the SoapySDRUtil tool.

If you think it's behaving the opposite, you're not alone. For SDRplay, the values may be "gain reduction", which means increasing them will actually provide lower signal levels. See also: https://github.com/pothosware/SoapySDRPlay/issues/60

ofadam, 2020-03-09:
This is helpful! Thanks.

ofadam, 2020-03-10:
I've tried the RFGR string, but it doesn't look like it does anything (and the browser immediately removes it from the URL after hitting return.) Am I using it wrong?

jketterl, 2020-03-11:
oh sorry, i missed your comment. it's not a URL parameter, it is intented to go into the `rf_gain` key of the profile in the `config_webrx.py`.

Example:

```
            "20m": {
                "name": "20m",
                "center_freq": 14150000,
                "rf_gain": "RFGR=2",
                "samp_rate": 500000,
                "start_freq": 14070000,
                "start_mod": "usb",
            },
```

ofadam, 2020-03-11:
Ah... so the just plain "gain" setting doesn't work at all? Or affects both
IF and RF?

jketterl, 2020-03-11:
no that setting won't do anything, it's not mapped in the code. If you pass a single value, e.g. `"rf_gain": 40` it will be up to the driver to distribute it across all the separate gains.

side note: I do intend to move all gain configs to the `gain` key, since I find `rf_gain` to be confusing. But that's for the future :)

closed 2020-03-21 by jketterl