jketterl/openwebrx #71: Start button in UI not doing anything

CountParadox, 2020-03-09:
Hey, 

Just finished installing everything to use rx_sdr with my hackrf,
Everything starts and I get no errors in logs, when I load up the web UI, i see the elements in the log correctly, but when I click the start button, nothing else happens. 

Please try for yourself: https://sdr.rei.moe

With no errors I'm at a loss what to check :(

jketterl, 2020-03-09:
I didn't get to see much, but that looked like an "old" OpenWebRX (i.e. the one from the ha7ilm repository that is discontinued). It's down now, so can't check in detail. That version did not support https URLs.

CountParadox, 2020-03-09:
Oh! That could be it... Hmm

CountParadox, 2020-03-09:
making progress, 

[image]

`lmartin@openwebrx:~$ openwebrx

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>

2020-03-09 22:47:12,407 - owrx.sdr - ERROR - The RTL source type "airspyhf" is not available. please check requirements.
2020-03-09 22:47:12,417 - owrx.sdr - ERROR - The RTL source type "sdrplay" is not available. please check requirements.
2020-03-09 22:47:12,417 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: HackRF One
Traceback (most recent call last):
  File "/usr/bin/openwebrx", line 11, in <module>
    load_entry_point('OpenWebRX==0.18', 'console_scripts', 'openwebrx')()
  File "/usr/lib/python3/dist-packages/owrx/__main__.py", line 54, in main
    server = ThreadedHttpServer(("0.0.0.0", pm.getPropertyValue("web_port")), RequestHandler)
  File "/usr/lib/python3.6/socketserver.py", line 456, in __init__
    self.server_bind()
  File "/usr/lib/python3.6/http/server.py", line 136, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.6/socketserver.py", line 470, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
lmartin@openwebrx:~$`

So I understand the address already in use part, as it seems to run itself on boot. But when I run it manually, it finds the HackRF One and loads it, but the web UI tells me that there is no SDR available.

jketterl, 2020-03-09:
Being "available" at this point only means that all the required programs to run it have been found. It may have failed in another way at a later point. I'm assuming it's a packaged install, so you should be able to get the logs with `sudo journalctl -u openwebrx`. You can also halt the instance started at boot with `sudo systemctl stop openwebrx`, but please be aware that running manually on the console will likely leak processes (most likely csdr, and in the case of HackRF, possibly hackrf_transfer).

CountParadox, 2020-03-09:
```
lmartin@openwebrx:/etc/openwebrx$ openwebrx

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>

2020-03-09 23:01:21,142 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: HackRF One
2020-03-09 23:01:21,620 - owrx.connection - DEBUG - client connection intitialized
2020-03-09 23:01:21,625 - owrx.source - DEBUG - activating profile 70cm
2020-03-09 23:01:21,629 - owrx.source - INFO - Started sdr source: hackrf_transfer -s 2400000 -g 30 -f 438800000 -r-|csdr convert_s8_f|nmux --bufsize 602112 --bufcnt 84 --port 48871 --address 127.0.0.1
Illegal instruction (core dumped)
Illegal instruction (core dumped)
2020-03-09 23:01:22,008 - owrx.source - DEBUG - shut down with RC=132
2020-03-09 23:01:22,032 - owrx.dsp - DEBUG - received STATE_FAILED, shutting down DspSource
```

jketterl, 2020-03-09:
that could be coming from either hackrf_transfer or, more likely, csdr... csdr is making use of the [AVX instruction set](https://en.wikipedia.org/wiki/Advanced_Vector_Extensions) to optimize performance. Unfortunately, that is unavailble on older CPUs.

CountParadox, 2020-03-09:
ooh! Maybe because its virtualised... let me try somethign

CountParadox, 2020-03-09:
Eureka!

you were spot on! 

It was using a virtualised kvm CPU, I switched it over to use the Host CPU and we're running!!

Thanks very much :)

closed 2020-03-09 by CountParadox

jketterl, 2020-03-09:
awesome :) enjoy :)