jketterl/openwebrx #73: Lime SDR not working

DB3DE, 2020-03-11:
Lime SDR seems not to work in 0.18.0 release (install via ready made raspy image)
I do not get any audio or waterfall. 

There is a double existence of a Soapy library (might be not related):
`root@limesdr:/home/pi# SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-4+b4
API Version: v0.6.0
ABI Version: v0.6
Install root: /usr
Search path: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6
Search path: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.6
Search path: /usr/local/lib/SoapySDR/modules0.6
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libHackRFSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libRedPitaya.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libairspySupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libaudioSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libbladeRFSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libosmosdrSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libremoteSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/librtlsdrSupport.so
Module found: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libuhdSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libPlutoSDRSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libairspyhfSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
Loading modules... [ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so)
  duplicate entry for lime (/usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.6/libLMS7Support.so)
done
Available factories...airspy, airspyhf, audio, bladerf, hackrf, lime, null, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd,
`

removal of the second library (rm /usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so solves the error above)

The SDR seems to be found:
`
pi@limesdr:/etc/openwebrx $ SoapySDRUtil --find="driver=lime"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Found device 0
  addr = 24607:1027
  driver = lime
  label = LimeNET-Micro [USB 2.0] 58399B3D2ED0
  media = USB 2.0
  module = FT601
  name = LimeNET-Micro
  serial = 0058399B3D2ED0
`

and the probe command gives details:
SoapySDRUtil --probe="driver=lime"

When started on commandline I get the following output:
pi@openwebrx:~ $ sudo -u openwebrx openwebrx |tee log.txt
2020-03-11 14:34:10,404 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: LiNET
2020-03-11 14:34:10,408 - owrx.source - DEBUG - activating profile 70cmup
2020-03-11 14:34:10,416 - owrx.source - INFO - Started sdr source: soapy_connector -s 4000000 -P 0 -g LNA=30 -d driver=lime -a LNAL -f 438100000 -p 50663 -c 45477
[INFO] Make connection: 'LimeNET-Micro [USB 2.0] 58399B3D2ED0'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: UNKNOWN
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M calibration values caching Disable
Setting gain element LNA: 30.000000 dB
IQ worker thread started
[... 126 lines cut ...]
2020-03-11 14:34:42,309 - owrx.dsp - DEBUG - adding new output of type secondary_fft
csdr tee: file opened: /tmp/openwebrx/openwebrx_pipe_1967510480_iqtee_pipe
2020-03-11 14:34:42,332 - owrx.dsp - DEBUG - adding new output of type wsjt_demod
csdr tee: file opened: /tmp/openwebrx/openwebrx_pipe_1967510480_iqtee2_pipe
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 151) + (input_size = 362) - 1
(overlap_length = 150) = taps_length - 1
csdr shift_addition_cc: reinitialized to -0.087
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
2020-03-11 14:34:42,344 - owrx.dsp - DEBUG - adding new output of type smeter
csdr bandpass_fir_fft_cc: filter initialized, low_cut = 0.0249, high_cut = 0.249
^C2020-03-11 14:35:26,198 - owrx.websocket - DEBUG - websocket loop ended; shutting down
signal 15 caught
2020-03-11 14:35:26,209 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
2020-03-11 14:35:26,212 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
2020-03-11 14:35:26,213 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
libusb: warning [libusb_exit] application left some devices open
2020-03-11 14:35:26,372 - owrx.source - DEBUG - shut down with RC=0
2020-03-11 14:35:26,375 - owrx.websocket - DEBUG - websocket loop ended; sending close frame
2020-03-11 14:35:26,377 - owrx.cpu - DEBUG - cpu usage thread shut down
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
BrokenPipeError: [Errno 32] Broken pipe
`
`

jketterl, 2020-03-11:
This may be caused by duplicate and conflicting library, there's also the chance that the packaged version is broken.

I think the SD image builds up until 2020-02-19 came with the self-compiled version only, so we know that one is working. I can remove the compilation from the build, would you mind testing the resulting image?

HuubH, 2020-03-11:
Experiencing the same problem. Happy to test the image.

jketterl, 2020-03-12:
For those that want to join testing: There's purpose-build RPi images in this folder: https://www.openwebrx.de/download/rpi.php?path=debugging

The one suffixed as -packaged uses the debian packages. I already have one report that this one seems to fail. It is also lacking udev rules.

The one suffixed as -compiles has the limesuite compiled from source.

HuubH, 2020-03-13:
Works for me with following config on a pi using the image    
"Lime": {
        "name": "LimeSDR-Mini",
        "type": "lime_sdr",
        "ppm": 0,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 30,
                "samp_rate": 2500000,
                "start_freq": 439275000,
                 "antenna": "Auto",
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 30,
                "samp_rate": 2500000,
                "antenna": "Auto",
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
            "Air": {
                "name": "Air",
                "center_freq": 130000000,
                "rf_gain": 30,
                "samp_rate": 5000000,
                "antenna": "Auto",
                "start_freq": 128000000,
                "start_mod": "am",

            },

Thanks a lot!

jketterl, 2020-03-13:
Which one did you pick? I'm assuming the `-compiled` one, right?

HuubH, 2020-03-13:
Correct, sorry should have mentioned that.

jketterl, 2020-03-21:
Alright, there's a new Image built today available on [the download page](https://www.openwebrx.de/download/rpi.php) - it doesn't contain any modifications from the `-compiled` image, but it's built using the official build process.

closed 2020-03-21 by jketterl