jketterl/openwebrx #74: NFM doesn`t work

UB0ALH, 2020-03-11:


jketterl, 2020-03-11:
Did you use a different device when listening a few hours ago? I am suspecting you may have an unsupported version of csdr installed.

UB0ALH, 2020-03-11:
No. I can`t do this - I have only one RTL-SDR (RTL2838).

jketterl, 2020-03-11:
Sorry I should have been more specific. I was referring to "device" as in client, so did you use a different browser, computer, smartphone or tablet?

UB0ALH, 2020-03-11:
))) I`m stupid fool!!! In Chrome it works 59 )))

I used Firefox before. Cleaning cache didn`t help - that should I do?

jketterl, 2020-03-11:
Try to install csdr from here: https://github.com/jketterl/csdr

your description of the problem plus the fact that you probably had an install of an older version of OpenWebRX on this machine means you're probably using an older version of csdr right now. While that is _mostly_ ok, it lacks the NFM deemphasis filter for 12kHz.

UB0ALH, 2020-03-11:
Ok. Thank  you very match !

closed 2020-03-11 by UB0ALH