jketterl/openwebrx #75: Path to JS

UB0ALH, 2020-03-11:
index.html:
"...
       <script src="static/openwebrx.js"></script>
        <script src="static/lib/jquery-3.2.1.min.js"></script>
        <script src="static/lib/jquery.nanoscroller.js"></script>
        <script src="static/lib/BookmarkBar.js"></script>
        <script src="static/lib/AudioEngine.js"></script>
        <script src="static/lib/ProgressBar.js"></script>
        <script src="static/lib/Measurement.js"></script>
        <script src="static/lib/FrequencyDisplay.js"></script>
        <link rel="stylesheet" type="text/css" href="static/lib/nanoscroller.css" />
        <link rel="stylesheet" type="text/css" href="static/css/openwebrx.css" />
..."

But in folder there is folder "static". All files are in "lib" folder. If I change "static/lib" on "lib" it doesn`t work at all. How it works?!

jketterl, 2020-03-11:
Well since this is an issue, the main thing is that it does work... So, does it work?

OpenWebRX isn't a regular webserver. It serves static files through the `/static` prefix. What comes after that (i.e. `lib/jquery-3.2.1.min.js`) relates to the "htdocs" folder in the checkout.

UB0ALH, 2020-03-11:
Thanks! I`ll search information for this item!

closed 2020-03-11 by UB0ALH