jketterl/openwebrx #76: Changes between "./openwebrx.py" & "sudo systemctl start openwebrx"?

UB0ALH, 2020-03-11:
If I use "sudo systemctl start openwebrx" I have page without warning message "Under construction".
If I use  "./openwebrx.py" I have page with this warning message.
This commands uses different index.html ?!

jketterl, 2020-03-11:
Well it seems you have two versions of OpenWebRX installed. While that isn't recommended, it's not a big problem. Just make sure you keep track of which one you're using.

I'm guessing you followed the repository installation and _additionally_ checked out the source code from here. Checking out the source code isn't really necessary, unless you want to do development, or have followed the manual installation method.

I hope you did not follow more than one installation method, because in that case your system probably has multiple installations of all the tools involved. That would be a big mess.

UB0ALH, 2020-03-11:
Hmm...
First I download  source code from here. I simply copy it to folder with ha7ilm version of openwebrx. It didnot work.

UB0ALH, 2020-03-11:
Thank you for such quick answer!!!

jketterl, 2020-03-11:
That doesn't surprise me, there's been a vast amount of changes since that version, and simply copying one folder over the other means that there's probably many old files in their old location plus new files in their new location.

The manual installation method does not have a simple "undo" method, that's why it's marked as "for advanced users only"...

I suggest you start with a new, clean checkout, and hope that the rest of the dependencies are in their place. You can of course keep the old folders as backups, just make sure you start from the right one.

Where did you get the systemd service from?

UB0ALH, 2020-03-11:
"Where did you get the systemd service from?" - I`ll find out it now... Thanks!

UB0ALH, 2020-03-11:
I found the source systemd service. Something was not good. On you project I go on link https://www.openwebrx.de/ , pressing OpenWebRX logo in upper-left corner. Here I found instructions https://www.openwebrx.de/download/debian.php.
That is all. This site have any attitude to you?

jketterl, 2020-03-11:
Yes, that's the new homepage. But for setting up a receiver, you should be following the documentation. Why didn't you stop after one set of instructions?

This is probably beyond repair... at this point I can only recommend to start from scratch :cry:

UB0ALH, 2020-03-11:
I repair my system. Thanks a lot!
I just want to explaine my mistake - maybe it will be useful for you?

closed 2020-03-11 by UB0ALH