jketterl/openwebrx #78: wsjtx-2.1.2

UB0ALH, 2020-03-11:
In yours instructions https://github.com/jketterl/openwebrx/wiki/Manual-Package-installation-(including-digital-voice) there is link to WSJT-X homepage - it doesn`t work.
Accordingly  wget http://physics.princeton.edu/pulsar/k1jt/wsjtx-2.1.2.tgz will have no effect.
I download this file from https://sourceforge.net/projects/wsjt/files/wsjtx-2.1.2/wsjtx-2.1.2.tgz/download

I wrote that, just you to know.

jketterl, 2020-03-11:
I just checked, and I'm confused, the link works for me, as well as the wget. This is the second time this has come up (see https://github.com/jketterl/openwebrx/issues/52#issuecomment-586738359). Is there some kind of blocking going on?

UB0ALH, 2020-03-11:
I don`t know about such kind of blocking (((
I planning to make your project adaptive to any kind of display. I tried it on mobile - it is very difficult & uncomfortable. SDR is "working" for many people in my Krasnoyarsk city.  And approximatly 50% of them use it on mobile phones & tablet.
As example you can see my first site http://арк-автогород.рф/ (http://xn----7sbbifg1b9abaohw.xn--p1ai/). This site made mostly for landscape,  but it works & in portrait landing.
If you are interesting, I can share you my results.

UB0ALH, 2020-03-11:
Hey! MidnightRAT - russian? )))
My coutry blocks some sites. For example, Russia tried to block Telegram. It still works good, but many site suffered from it...

UB0ALH, 2020-03-11:
MidnightRAT can live in China - too many blocks of sites?

UB0ALH, 2020-03-11:
I thought - it is not blocking for Russia 100%.
Then site is blocked we see message about it. In this case I see only timeout!

UB0ALH, 2020-03-11:
traceroute http://physics.princeton.edu 
says "Unknown  name or service"

jketterl, 2020-03-11:
I don't know exactly what's happening there. Maybe someone else can check the links. I have been downloading from that site many times in the past, and there's also automatic builds that download from there. It's the official download site, at least to my knowledge.

As for the mobile usage: Yes, that is one of the issues that I still need to address. I haven't had time to overhaul the frontend yet, so much of it still uses the original layout, which has never been optimized for mobile. To make it really mobile friendly, I may need to rebuild it completely.

The new homepage https://www.openwebrx.de/ has been built with mobile in mind from the start.

try `traceroute physics.princeton.edu` (no http)

```
root@de9cebbc1303:/# traceroute -n physics.princeton.edu
traceroute to physics.princeton.edu (128.112.100.6), 30 hops max, 60 byte packets
 1  172.17.0.1  0.065 ms  0.029 ms  0.023 ms
 2  192.168.1.1  0.328 ms  0.271 ms  0.208 ms
 3  192.168.178.1  0.552 ms  0.604 ms  0.559 ms
 4  62.245.142.157  15.355 ms  15.440 ms  18.740 ms
 5  62.245.142.155  21.281 ms  21.239 ms  21.179 ms
 6  212.18.6.109  18.725 ms 212.18.7.63  18.454 ms 212.18.6.109  20.767 ms
 7  80.81.202.30  35.833 ms  35.981 ms  35.693 ms
 8  184.105.223.109  28.742 ms  28.664 ms  28.325 ms
 9  72.52.92.85  30.633 ms  23.395 ms  23.240 ms
10  184.105.80.37  35.356 ms  35.216 ms  34.919 ms
11  72.52.92.166  102.737 ms  104.629 ms  102.700 ms
12  216.66.49.74  107.931 ms  105.353 ms  105.402 ms
13  128.112.12.66  107.323 ms  103.200 ms  101.747 ms
14  128.112.100.6  103.534 ms  103.648 ms  105.434 ms
```

UB0ALH, 2020-03-11:
https://website.informer.com/physics.princeton.edu doesn`t know Whois...

UB0ALH, 2020-03-11:
tolpa@Tolpa:~$ traceroute -n physics.princeton.edu
traceroute to physics.princeton.edu (128.112.100.6), 30 hops max, 60 byte packets
 1  192.168.1.1  0.266 ms  0.281 ms  0.306 ms
 2  172.27.192.254  1.177 ms  1.184 ms  1.252 ms
 3  94.73.254.195  1.746 ms  1.957 ms  2.176 ms
 4  94.73.254.223  1.510 ms  1.596 ms  1.778 ms
 5  94.73.254.201  1.537 ms  1.578 ms  1.804 ms
 6  109.226.254.253  2.129 ms  1.883 ms  1.986 ms
 7  37.29.24.5  2.792 ms 188.170.166.97  1.286 ms 37.29.24.5  1.816 ms
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

UB0ALH, 2020-03-11:
"To make it really mobile friendly, I may need to rebuild it completely." - I don`t think so!

UB0ALH, 2020-03-11:
I think a lot of thing could be resolved just with css.

UB0ALH, 2020-03-11:
[image]

UB0ALH, 2020-03-11:
Timeout (((

UB0ALH, 2020-03-11:
tolpa@Tolpa:~$ wget http://physics.princeton.edu/pulsar/k1jt/wsjtx-2.1.2.tgz
--2020-03-12 05:09:56--  http://physics.princeton.edu/pulsar/k1jt/wsjtx-2.1.2.tgz
Распознаётся physics.princeton.edu (physics.princeton.edu)… 128.112.100.6
Подключение к physics.princeton.edu (physics.princeton.edu)|128.112.100.6|:80... ошибка: Время ожидания соединения истекло.
Продолжение попыток.

--2020-03-12 05:12:08--  (попытка: 2)  http://physics.princeton.edu/pulsar/k1jt/wsjtx-2.1.2.tgz
Подключение к physics.princeton.edu (physics.princeton.edu)|128.112.100.6|:80... ошибка: Время ожидания соединения истекло.
Продолжение попыток.

--2020-03-12 05:14:20--  (попытка: 3)  http://physics.princeton.edu/pulsar/k1jt/wsjtx-2.1.2.tgz
Подключение к physics.princeton.edu (physics.princeton.edu)|128.112.100.6|:80...

UB0ALH, 2020-03-11:
Third attempt to connect (((

UB0ALH, 2020-03-11:
princeton.edu opens easily

UB0ALH, 2020-03-11:
[image]

UB0ALH, 2020-03-11:
tolpa@Tolpa:~$ traceroute -n princeton.edu
traceroute to princeton.edu (140.180.223.22), 30 hops max, 60 byte packets
 1  192.168.1.1  0.279 ms  0.278 ms  0.301 ms
 2  172.27.192.254  1.248 ms  1.261 ms  1.375 ms
 3  94.73.254.193  1.787 ms  1.985 ms  2.040 ms
 4  94.73.254.89  1.367 ms  1.665 ms  1.745 ms
 5  94.73.254.201  1.540 ms  1.682 ms  1.866 ms
 6  109.226.254.253  1.888 ms  1.850 ms  2.053 ms
 7  37.29.24.5  4.729 ms 188.170.166.97  1.543 ms  1.527 ms
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

jketterl, 2020-03-11:
OK I'm about to go to bed, so let me just wrap this up. There's nothing I can do about the fact that that website is unavailable to you. The website appears to be available (https://downforeveryoneorjustme.com/physics.princeton.edu) and since I'm neither the operator of that site, nor an ISP technician, I cannot do anything to analyze the problem further, or fix it. There is absolutely no point in spamming it over and over, so stop it now.

As for the mobile view: That's nothing I am working on right now. There's more important issues I should be working on, but unfortunately I'm currently spending all my time answering E-Mails and tickets, so the project development is completely stalled at the moment.

UB0ALH, 2020-03-11:
((( I understand. Thanks for help & good luck!!!

closed 2020-03-11 by UB0ALH