jketterl/openwebrx #79: missing required dependencies but have

MidnightRAT, 2020-03-11:
What wrong in my config?

I have read the out:
"you are missing required dependencies to run openwebrx. please check that the following core requirements are installed:
csdr, nmux, nc"

But i have this:
[root@rat openwebrx]# which csdr
/usr/bin/csdr
[root@rat openwebrx]# which nmux
/usr/bin/nmux
[root@rat openwebrx]# which nc
/usr/bin/nc

UB0ALH, 2020-03-11:
Try to reboot

jketterl, 2020-03-12:
The dependencies are there, I can think of a few reasons why it may still be fail:

- OpenWebRX needs to find those dependendies via the `PATH` environment variable. How are you starting it and could it be possible that the variable is lost somewhere?
- The dependencies must not only be there, they must be runnable. So please check if the following commands give any errors:
  - `csdr`
  - `nmux --help`
  - `nc --help`
- This might also be a simple permission problem, the user that OpenWebRX runs as must be able to execute the binaries, so please check the permissions on the files.

MidnightRAT, 2020-03-12:
Please check it

Thanks.

[root@rat openwebrx]# ./openwebrx.py 
OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________
Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>
you are missing required dependencies to run openwebrx. please check that the following core requirements are installed:
csdr, nmux, nc
[root@rat openwebrx]# whoami 
root
[root@rat openwebrx]# env | grep -i path
CUDA_INC_PATH=/usr/include/cuda
MOZ_GMP_PATH=/usr/lib64/mozilla/plugins/gmp-gmpopenh264/system-installed
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/root/bin
[root@rat openwebrx]# csdr --help
csdr - a simple commandline tool for Software Defined Radio receiver DSP.

usage: 

    csdr function_name <function_param1> <function_param2> [optional_param] ...

list of functions:

    convert_u8_f
    convert_f_u8
    convert_s8_f
    convert_f_s8
    convert_f_s16
    convert_s16_f
    convert_f_s24 [--bigendian]
    convert_s24_f [--bigendian]
    realpart_cf
    clipdetect_ff
    limit_ff [max_amplitude]
    gain_ff <gain>
    clone
    none
    yes_f <to_repeat> [buf_times]
    detect_nan_ff
    dump_f
    shift_math_cc <rate>
    shift_math_cc --fifo <fifo_path>
    shift_addition_cc <rate>
    shift_addition_cc --fifo <fifo_path>
    shift_addition_cc_test
    shift_table_cc <rate> [table_size]
    shift_addition_fc <rate>
    shift_addition_fc --fifo <fifo_path>
    decimating_shift_addition_cc <rate> [decimation]
    dcblock_ff
    fastdcblock_ff
    fmdemod_atan_cf
    fmdemod_quadri_cf
    fmdemod_quadri_novect_cf
    deemphasis_wfm_ff <sample_rate> <tau>
    deemphasis_nfm_ff <one_of_the_predefined_sample_rates>
    amdemod_cf
    amdemod_estimator_cf
    fir_decimate_cc <decimation_factor> [transition_bw [window]]
    fir_interpolate_cc <interpolation_factor> [transition_bw [window]]
    firdes_lowpass_f <cutoff_rate> <length> [window [--octave]]
    firdes_bandpass_c <low_cut> <high_cut> <length> [window [--octave]]
    agc_ff [hang_time [reference [attack_rate [decay_rate [max_gain [attack_wait [filter_alpha]]]]]]]
    fastagc_ff [block_size [reference]]
    rational_resampler_ff <interpolation> <decimation> [transition_bw [window]]
    old_fractional_decimator_ff <decimation_rate> [transition_bw [window]]
    fractional_decimator_ff <decimation_rate> [num_poly_points ( [transition_bw [window]] | --prefilter )]
    fft_cc <fft_size> <out_of_every_n_samples> [window [--octave] [--benchmark]]
[... 112 lines cut ...]
  -z                         Zero-I/O mode, report connection status only
      --append-output        Append rather than clobber specified output files
      --send-only            Only send data, ignoring received; quit on EOF
      --recv-only            Only receive data, never send anything
      --no-shutdown          Continue half-duplex when receiving EOF on stdin
      --allow                Allow only given hosts to connect to Ncat
      --allowfile            A file of hosts allowed to connect to Ncat
      --deny                 Deny given hosts from connecting to Ncat
      --denyfile             A file of hosts denied from connecting to Ncat
      --broker               Enable Ncat's connection brokering mode
      --chat                 Start a simple Ncat chat server
      --proxy <addr[:port]>  Specify address of host to proxy through
      --proxy-type <type>    Specify proxy type ("http", "socks4", "socks5")
      --proxy-auth <auth>    Authenticate with HTTP or SOCKS proxy server
      --proxy-dns <type>     Specify where to resolve proxy destination
      --ssl                  Connect or listen with SSL
      --ssl-cert             Specify SSL certificate file (PEM) for listening
      --ssl-key              Specify SSL private key (PEM) for listening
      --ssl-verify           Verify trust and domain name of certificates
      --ssl-trustfile        PEM file containing trusted SSL certificates
      --ssl-ciphers          Cipherlist containing SSL ciphers to use
      --ssl-alpn             ALPN protocol list to use.
      --version              Display Ncat's version information and exit

See the ncat(1) manpage for full options, descriptions and usage examples

MidnightRAT, 2020-03-13:
Original version is sturt up

`
[root@rat opt]# git clone https://github.com/ha7ilm/openwebrx.git
Клонирование в «openwebrx»…
remote: Enumerating objects: 1038, done.
remote: Total 1038 (delta 0), reused 0 (delta 0), pack-reused 1038
Получение объектов: 100% (1038/1038), 6.45 MiB | 3.70 MiB/s, готово.
Определение изменений: 100% (639/639), готово.
[root@rat opt]# 
[root@rat openwebrx]# ./openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-main] nmux_bufsize = 602112, nmux_bufcnt = 84
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2400000 -f 144250000 -p 0 -g 5 -| nmux --bufsize 602112 --bufcnt 84 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0xc912e0: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2867 | csdr logaveragepower_cf -70 4096 93 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
[openwebrx-main] Starting HTTP server.
Ncat: Version 7.80 ( https://nmap.org/ncat )
Ncat: Connected to 127.0.0.1:4951.
nmux: pthread_create() done, clients now: 2
client 0xc91c20: started!
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 144250000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
Allocating 15 zero-copy buffers
client 0xc912e0: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32192.168.247.200 - - [13/Mar/2020 02:01:10] "GET / HTTP/1.1" 200 -
[openwebrx-httpd] client 192.168.247.200#adfba3e44f41c2da39ef357b8d457771 :: client added. Clients now: 1
192.168.247.200 - - [13/Mar/2020 02:01:10] "GET /sdr.js HTTP/1.1" 200 -

`

jketterl, 2020-03-13:
OK, that seems to be fine. Starting to wonder a little what else could be going wrong.

What's your python version? `python3 --version`

And would you mind checking the path that is configured as `temporary_directory`? I have a suspicion there might be the same Exception as if one of the dependencies is missing if that isn't there.

MidnightRAT, 2020-03-13:
[root@rat openwebrx]# python --version
Python 3.7.6
[root@rat openwebrx]# python3 <Tab>
python3            python3.7          python3.7m         python3-html2text  
[root@rat openwebrx]# python3 --version
Python 3.7.6

MidnightRAT, 2020-03-13:
how can i see tmp variable for my python
All __pycache__ folders and *.pyc files was be deleted before clean start
/tmp folder was clean and OS reboooted
[root@rat New.openwebrx]# uname -a
Linux rat 5.5.8-200.fc31.x86_64 #1 SMP Thu Mar 5 21:28:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
[root@rat New.openwebrx]# cat /etc/redhat-release 
Fedora release 31 (Thirty One)
[root@rat New.openwebrx]# pwd
/opt/New.openwebrx

jketterl, 2020-03-13:
It's not a python variable, it's in `config_webrx.py` (or `/etc/openwebrx/config_webrx.py` if it exists)

MidnightRAT, 2020-03-13:
Your are right. Thank You very and very mach.
config was get from docker
temporary_directory = "/tmp/openwebrx"
/tmp/openwebrx was absent

docker cant find device after reboot and i return to native app
Thanks

closed 2020-03-13 by MidnightRAT

jketterl, 2020-03-13:
Thanks for your feedback. I have followed this up with #82 since the behaviour may affect other users in the future and isn't easy to diagnose. I'd like to fix this.

Docker: I have had a report this week that USB devices may be unavailable inside docker containers if the host has loaded the corresponding kernel modules. Sounds like you may have hit that issue as well. I've added documentation in the wiki: https://github.com/jketterl/openwebrx/wiki/Getting-Started-using-Docker