jketterl/openwebrx #81: Unable to set negative coordinate for "receiver_gps"

ofadam, 2020-03-13:
I'm in North American and even though I've set my GPS coordinates correctly in the config file (40, -89), it still places me on the other side of the world. Do I need to specify negative coordinates in a special way?

jketterl, 2020-03-13:
Well, let me start with "it _should_ work that way". I just tried to reproduce it, but I'm not sure where to click. What exact feature are you referring to?

I just tried clicking the "maps" link in the receiver description, as well as the builtin map - they both send me to a location in the US when I use your coordinates. That seems reasonable to me.

ofadam, 2020-03-15:
In the config file, this line:

receiver_gps = (47.000000, 19.000000)

When I put in (40.xxxxx,-89.xxxxxx), it ignores the "dash" and maps it in Asia instead of North America.

jketterl, 2020-03-15:
I just tried this:

`receiver_gps = (40.1234,-89.1234)`

and this is the result when I click the maps link (with both 0.18.0 and current develop):

[image]

again: please be more specific. I have understand your expectations, but I still need to understand what interaction with the receiver to look at.

ofadam, 2020-03-15:
Sorry, I'm doing a poor job of this!

With those coordinates configured, I set up OpenwebRX to decode APRS on 2m.
The reporting to aprs.fi works fine, but is reporting my station as on the
other side of the world.

https://aprs.fi/#!mt=roadmap&z=11&call=a%2FKD9NYE&timerange=3600&tail=3600

jketterl, 2020-03-15:
Thank you very much. The `receiver_gps` is used in a number of places, so it was a little ambigious before.

With your latest comment I was able to locate the problem: The logic in the direwolf config was generating a negative west coordinate, which kind of cancels itself out. I'm also impressed it was accepted by the network.

I think I fixed it now, are you able to switch to the latest development version to test it?

ofadam, 2020-03-15:
That did the trick! Thanks so much for pushing ahead with me!

jketterl, 2020-03-15:
No Problem :) I do want those bugs out, too! :grin:

closed 2020-03-15 by jketterl