jketterl/openwebrx #83: Adaptive landing

UB0ALH, 2020-03-13:
http://94.73.222.226:8073 - adaptive landing in process...
Stopped on DMR Timeslotes (((
Can`t adapt picture of user, group & muted. Can you help?

jketterl, 2020-03-13:
I don't know, how am I supposed to help?
The size seems to be set on `.openwebrx-meta-slot`.
Are you trying to simply scale everything down on smaller devices?

UB0ALH, 2020-03-13:
.openwebrx-meta-slot - Yes. I understand that.  
"Are you trying to simply scale everything down on smaller devices?" - yes. It bad? You can see portrait & landscape on mobile phone... It works. At least I see now all on screen. 
I can`t scale pictures of User, Group & Mute (((
I can scale "openwebrx-meta-slot", but can`t do anything with this pictures (((

jketterl, 2020-03-13:
the pictures scale with their containers... so they should respond to whatever you do with `.openwebrx-meta-slot`. Not sure what you're trying to achieve...

Well, I'm concerned because it will become pretty much unreadable on a phone, for example. I'm not convinced that this will result in a good user experience...

UB0ALH, 2020-03-13:
I don`t see good-look  mobile-sdr (((
& I have no idea how it shoud be... - You have an idea?!
On scalable I can use bookmarks, can use volume & squelch, I can load-unload log, tuner & etc... For me it is good minimum.

UB0ALH, 2020-03-13:
But your are right - it  is not a good user experience!

UB0ALH, 2020-03-13:
I scaled .openwebrx-meta-slot
Pictures not keep it scale

jketterl, 2020-03-13:
well, the basic problem is that an sdr need space, which is probably why there's no good solution. that's also why i put this up for later, i wanted to think this through thoroghly and maybe do some experimenting before changing up stuff.

I see what you mean now. I don't know why it worked when I tried before. You'll need to set `background-size: contain` on the `openwebrx-meta-user-image`. There's also a static height on that element that needs to be set to something else (don't have any suggestions for that right now).

UB0ALH, 2020-03-13:
Thanks, Jakob!
Yes an sdr needs space (((
I thought about good user experience. (You infected me!)))
Really I`v got one idea. I tried it tomorrow & if it works, I show it.

jketterl, 2020-03-13:
before working on a dedicated mobile view i'd also really like to re-work the client-server protocol in a more demand-oriented way. right now, the data that is sent is pretty much fixed, whether or not you want to actually use it. so basically, if i was to build a frontend without a waterfall, i'd still be receiving waterfall data, even though i'm not using it. basically, i'd like to have this scale with the application. if an element isn't visible, disable it's backend code as well. it's not only mobile stuff that will profit that way... but it should all come together at some point.

UB0ALH, 2020-03-13:
I understand you! Of course waterfall-data & other data should be transferred even we don`t want to see it or use it.
Idea I keep in mind - mobile interface having nothing common with desktop ;)

jketterl, 2020-03-13:
well that doesn't necessarily mean it won't have nothing in common with the desktop. when designed right, it will employ responsive technology to show whatever it needs to show. but the idea is that when elements are hidden, they will be disabled completely, and individually.

I don't want to build a second endpoint and have that deliver exactly what the mobile view needs, because depending on the device and orientation, what is really needed will vary.

UB0ALH, 2020-03-13:
"they will be disabled completely, and individually" - YES! )))
Rebuild - revolution. It`s no good & tiresomely (((
The idea is to present data for desktop another way the  for mobile.

UB0ALH, 2020-03-13:
Your help about "background-size: contain" & "openwebrx-meta-slot.muted:before" worked on 50%.
Thanks. Lets leave it for some time. I think now another way...
[image]
[image]

jketterl, 2020-03-13:
you may need to re-style the box content with flexbox. it wasn't designed with such flexibility in mind :wink:

closed 2020-05-03 by jketterl