jketterl/openwebrx #85: Question Public SDR List on https://www.openwebrx.de?

linuxonlinehelp, 2020-03-16:
Since sdr.hu is closed ONLY for HAMS with a real CALLSIGN the DX Users are NOT able to register or to listen. Thats is not FAIR in my eyes it prevents all upcomming hams or young people to join and to learn "hearing the Signs-Locators"

Should we not open a PUBLIC List of all openwebrx Radios for all humans without borders? 
That may be great!

amontefusco, 2020-03-16:
I agree. 
Ad besides, who is maintaining that site now?

jketterl, 2020-03-16:
I agree with you, and I don't really understand the reasoning of the callsign requirement. But, building a new index is quite a bit of work, and I haven't gotten round to it yet. There's a lot of work on OpenWebRX itself as it is right now :wink: 

Also, I do have a few ideas on how to improve things. Mostly, I'd like to build an open index, i.e. not limited to OpenWebRX, so it'll probably not be on openwebrx.de.

amontefusco, 2020-03-16:
> I agree with you, and I don't really understand the reason for the callsign requirement.

Maybe because it guarantees the uniqueness of the primary key :-)

linuxonlinehelp, 2020-05-17:
purged

closed 2020-05-17 by linuxonlinehelp