jketterl/openwebrx #89: [INFO] How to increase in a safe way the rx frequency in openwebrx.js

amontefusco, 2020-03-17:
There is a method or other clean way to increase the rx frequency with a small offset ?

I'm trying to add two buttons on the top bar (where Log, Status, etc are) to allow to fine-tune even when a keyboard or mouse are not available (tablet/phones).

I added a couple of items with their own onclick function but I don't know how to change the frequency int the model.

```
            <section id="openwebrx-main-buttons">
                <ul>
                   <li data-inc-freq="openwebrx-panel-incfreq" onclick="inc_freq_hz (+100);"><img src="static/gfx/openwebrx-panel-status.png" /><br/>Incr  freq</li>
                    <li data-inc-freq="openwebrx-panel-decfreq" onclick="inc_freq_hz (-100);"><img src="static/gfx/openwebrx-panel-status.png" /><br/>Dec freq</li>
                    <li data-toggle-panel="openwebrx-panel-status"><img src="static/gfx/openwebrx-panel-status.png" /><br/>Status</li>
                    <li data-toggle-panel="openwebrx-panel-log"><img  src="static/gfx/openwebrx-panel-log.png" /><br/>Log</li>
                    <li data-toggle-panel="openwebrx-panel-receiver"><img src="static/gfx/openwebrx-panel-receiver.png" /><br/>Receiver</li>
                    <li><a href="map" target="_blank"><img src="static/gfx/openwebrx-panel-map.png" /><br/>Map</a></li>
                    <li><a href="admin" target="_blank"><img src="static/gfx/openwebrx-panel-settings.png" /><br/>Settings</a></li>
```

amontefusco, 2020-03-17:
Changing my event handler as follows

```
function inc_freq_hz (ofs) {
   console.log("################## %d %d", demodulators[0].get_offset_frequency() + center_freq, ofs);  
   var newf = demodulators[0].get_offset_frequency() + center_freq + ofs;
   tunedFrequencyDisplay.setFrequency(newf);  
   demodulators[0].set_offset_frequency(demodulators[0].get_offset_frequency() + ofs) 
}

```

it seems to work now.
I doubt it is the right way though!

jketterl, 2020-03-18:
I haven't had a chance to look at it myself, too many headaches at the moment (both literal and figurative). The code you're working with right there is still more or less untouched by me, but I can tell you that I'm facing the same troubles, and I do intend to rewrite it sooner or later. It doesn't really have a straightforward interfaces, and I usually end up calling more methods than I thought at the beginning.

When adding buttons to the header, please keep in mind that somewhere in there is a method that will hide some items depending on the available space. This is just from the top of my head right now, so no idea if your code will be affected.

jketterl, 2020-05-02:
alright... the demodulator rewrite is coming along, the gears are now aligned better and there's simpler APIs that will do the work for you.

To obtain the demodulator, it's still necessary to do some jquery magic... it may seem a bit overkill, but given the design of the frontend, this allows proper wrapping of all the logic.

Here's an example of how you can modify the frequency:

```
var demod = $('#openwebrx-panel-receiver').demodulatorPanel().getDemodulator();
demod.set_offset_frequency(demod.get_offset_frequency() + 10000);
```

Please make sure you don't store the `demod` variable for long-term use as a new instance is created on every mode change.

It's also possible to implement `increment()` / `decrement()` methods on the demodulator itself. There's nothing that would require them now, so they're not available out of the box.

jketterl, 2020-05-24:
any updates on this? I'd like to get this closed.

amontefusco, 2020-05-24:
Still experimenting, due to my poor skill in JS.

closed 2020-08-26 by jketterl