jketterl/openwebrx #90: digital_voice_unvoiced_quality

UB0ALH, 2020-03-20:
Hello!
By default digital_voice_unvoiced_quality=1.
In comment you say it can be up to 3B+.
I made default digital_voice_unvoiced_quality=3B+, but the quality of voice in dmr seems had no any changes...
I made something wrong?

jketterl, 2020-03-20:
The comment read:s
```
if you're running on a Raspi (up to 3B+) you'll want to leave this on 1
```
where the 3B+ is referring to the model of the Rasbperry Pi. The 3B+ has a hard time handling anything higher, and the comment is there to tell you to leave the value at 1.

This is supposed to be a number, so setting it to `digital_voice_unvoiced_quality=3B+` should give you syntax errors.

UB0ALH, 2020-03-20:
))) thanks! funny....

closed 2020-03-20 by UB0ALH

jketterl, 2020-03-20:
don't think you're missing much. in my testing it was hard to hear a difference at all. I wouldn't be able to tell the difference unless somebody would play me the same audio in a side-by-side comparison. My guess is that the codec implementation (mbelib) would need overall improvement to get better results.