jketterl/openwebrx #91: squelch_level

UB0ALH, 2020-03-21:
Can I set up the default squelch_level in start up?

jketterl, 2020-03-21:
Yes you can :) All you need to set is the `initial_squelch_level`. Should work per device and per profile, but as always: if you want to use it on a profile, all profiles will need to have that setting.

BTW, I think it's time you head over to our mailing list https://groups.io/g/openwebrx - much better place to ask questions. Issues should be things that don't work as intended :)

UB0ALH, 2020-03-22:
sorry! ok )))
i should mail to openwebrx@groups.io ?

jketterl, 2020-03-25:
I think you'll have to subscribe before any messages are accepted via E-Mail. I'm using the website for all my posting, so I'm not too familiar with the mailing stuff.

jketterl, 2020-03-25:
Btw: if you happen to have discord, you can also join us there :) https://discord.gg/SxwEhke

jketterl, 2020-04-19:
is this issue still ongoing, or can we close it up? just asking because there's never been any feeback if the `initial_squelch_level` works :wink:

closed 2020-05-03 by jketterl